wadpac/GGIR #226: Axis lable issue in part 4

DianeU, 2019-09-17:
Hi, 
I ran the following code on multiple directories, and I keep getting the same error in one of the directories. What can I do to trouble shoo the error better? Sorry that this is not exactly a minimal example, but I am not an expert in GGIR, and would just like to know what I can do to understand the problem better. Thank you very much. Sorry I am also having trouble with the editor.

`
require(GGIR);
require(stringi);
require(randomForest);
require(data.table);
require(signal);
require(zoo)
`
`datadir <- "M:/Big CSV Files/01_data"
  outputdir="M:/accelerometry/03_results"
  g.shell.GGIR(

						 mode=1:5,
						 datadir=datadir,
						 outputdir=outputdir,
						 f0=1,
						 f1=c(),
						 do.imp=T, 
						 overwrite=FALSE,
						 # Part 1 parameters: #-------------------------------
						 windowsizes = c(5,900,3600),
						 do.cal=TRUE, 
						 do.enmo = TRUE,
						 do.hfen=TRUE,
						 do.hfenplus=TRUE,
						 do.anglex=TRUE, 
						 do.angley=TRUE, 
						 anglez=TRUE, 
						 do.dev_roll_med_acc_x =TRUE, 
						 do.dev_roll_med_acc_y =TRUE, 
						 do.dev_roll_med_acc_z =TRUE, 
						 chunksize=1, 
						 # Part 2 parameters: #-------------------------------
						 strategy = 3, 
						 ndayswindow=7,
						 hrs.del.start = 1, 
						 hrs.del.end = 1, 
						 ndayswindow=8,
						 includedaycrit = 16, 
						 L5M5window = c(0,24), 
						 M5L5res = 10, 
						 winhr = c(1,5,10), 
						 qlevels = c(c(1380/1440),c(1410/1440)), 
						 qwindow=c(0,24), 
						 ilevels = c(seq(0,400,by=25),8000), 
						 mvpathreshold =c(100,120),
						 boutcritr =0.8,
						 printsummary=T,
						 mvpadur = c(1,5,10),
						 # Part 3 parameters: #------------------------------- 
						 timethreshold= c(5,10),
						 anglethreshold=5, 
						 ignorenonwear = TRUE,
						 # Part 4 parameters: #------------------------------- 
						excludefirstlast = TRUE, 
						 includenightcrit = 16,
						 def.noc.sleep = c(), 
						 #loglocation= "D:/sleeplog.csv",
						 outliers.only = FALSE, 
						 criterror = 4, 
						 relyonsleeplog = FALSE, 
						 def.noc.sleep = c(21,9),
						 do.visual = TRUE,
						 nnights = 9, 
						 #Part 5 parameters: #------------------------------- 
						 # Key functions: Merging physical activity with sleep analyses 
						 threshold.lig = c(40, 50),
						 threshold.mod = c(100),
						 threshold.vig = c(400),
						 boutcriter = 0.8, 
						 boutcriter.in = 0.9, 
						 boutcriter.lig = 0.9, 
						 boutcriter.mvpa = 0.8, 
						 do.EMNO=T,
						 boutdur.in = c(20),
						 boutdur.lig = c(10), 
						 boutdur.mvpa = c(1,10), 
						 timewindow = c("WW"), 
						 # Report generation #------------------------------- 
						 do.report=c(2,4,5))`

Last bit of the output:

`Part 4
 1 2 3 NEW  4 5 6 NEW  7 8 9 NEW  10 11 12 NEW  13 14 NEW  15 16 17 NEW  18 19 N
EW  20 21 22 NEW  23 24 NEW  25 26 27 28 NEW  29 30 31 NEW  32 33 34 NEW  35 36
37 NEW  38 39 40 41 NEW  42 43 44 45 NEW  46 47 48 NEW  49 50 51 NEW  52 53 54 5
5 NEW  56 57 NEW  58 59 60 61 NEW  62 63 64 NEW  65 66 67 NEW  68 69 70 71 NEW
72 73 74 75 NEW  76 77 78 NEW  79 80 81 NEW  82 83 84 85 86 NEW  87 88 89 NEW  9
0 91 92 NEW  93 94 95 96 NEW  97 98 99 100 101 NEW  102 103 104 NEW  105 106 107
 108 109 NEW  110 111 112 113 NEW  114 115 116 NEW  117 118 119 120 NEW  121 122
 123 NEW  124 125 126 NEW  127 128 129 NEW  130 131 132 NEW  133 134 135 136 137
 NEW  138 139 140 141 NEW  142 143 144 NEW  145 146 147 148 NEW  149 150 151 152
 NEW  153 154 NEW  155 156 157 NEW  158 159 160 161 162 163 164 NEW  165 166 167
 NEW  168 169 170 NEW  171 172 173 NEW  174 175 176 NEW  177 178 NEW  179 180 18
1 182 183 NEW  184 185 186 187 188 189 190 191 192 193 194 195 196 197 198 199 2
00 201 202 203 204 205 206 207 208 209 210 211 212 213 214 215 216 217 218 219 2
20 221 222 223 224 225 226 227 228 229 230 231 232 233 234 235 236 237 238 239 2
40 241 242 243 244 245 246 247 248 249 250 251 252 253 254 255 256 257 258 259 2
60 261 262 263 264 265 266 267 268 269 270 271 272 273 274 275 276 277 278 279 2
80 281 282 283 284 285 286 287 288 289 290 291 292 293 294 295 296 297 298 299 3
00 301 302 303 304 305 306 307 308 309 310 311 312 313 314 315 316 317 318 319 3
20 321 322 323 324 325 326 327 328 329 330 331 332 333 334 335 336 337 338 339 3
40 341 342 343 344 345 346 347 348 349 350 351 352 353 354 355 356 357 358 359 3
60 361 362 363 364 365 366 367 368 369 370 371 372 373 374 375 376 377 378 379 3
80 381 382 383 384 385 386 387 388 389 390 391 392 393 394 395 396 397 398 399 4
00 401 402 403 404 405 406 407 408 409 410 411 412 413 414 415 416 417 418 419 4
20 421 422 423 424 425 426 427 428 429 430 431 432 433 434 435 436 437 438 439 4
40 441 442 443 444 445 446 447 448 449 450 451 452 453 454 455 456 457 458 459 4
60 461 462 463 464 465 466 467 468 469 470 471 472 473 474 475 476 477 478 479 4
80 481 482 483 484 485 486 487 488 489 490 491 492 493 494 495 496 497 498 499 5
00Error in axis(side = 2, at = 1:nnpp, labels = idlabels, las = 1, cex.axis = 0.
5) :
  'at' and 'labels' lengths differ, 40 != 4027
Calls: g.shell.GGIR -> g.part4 -> axis
Execution halted`

vincentvanhees, 2019-09-18:
1. Set do.visual = FALSE, this will turn off the creation of the visualisation and should avoid the error. Check that the rest of GGIR, e.g. the part 4 reports are created and look plausible.

2. Next step, set do.visual =TRUE, and outliers.only = TRUE. This will make that only nights with a large discrepancy between sleep period time estimate and onset/waking up time are plotted.

3. If it still work then that indicates that GGIR is somehow not able to create this many pages in the visualisation.... something I may be able to investigate with one of the datasets I have here.

Note: def.noc.sleep = c() will tell part4 to use the somewhat naive sleep period time detection algorithm. Setting it to def.noc.sleep = 1 or commenting it out will make that the algortihm I published last year is used. At the moment you are specifying def.noc.sleep twice.

vincentvanhees, 2020-04-15:
I have now added a few extra checks before the labels are created, which may address the issue. If the issue persists then please let me know.

closed 2020-04-15 by vincentvanhees