wadpac/GGIR #231: visual report improvement idea

vincentvanhees, 2019-09-25:
The visual reports that are produced at the very end of g.shell.GGIR do not distinguish between daytime sustained inactivity and nocturnal sustained inactivity (sleep). Possibly it is not too much work to make this distinction with colours.

m-patterson, 2019-09-25:
I can work on some possible solutions and send them out to the group to see what people would like to see. I was also thinking of adding light PA as well. I don't have any sample data with a light sensor in it and it seems that light sensors aren't used that much, so I was thinking of leaving this information out.

m-patterson, 2019-10-07:
option to create new plots. I've kept the old plots available in case any users prefer them. Users adjust the 'visualreport' input variable to set which plot they want to see (or both, or none)
no plot: visualreport = 0
original plot: visualreport = 1
new plot: visualreport = 2
both plots: visualreport = c(1,2)

Here is what the new plot looks like:
[image]

m-patterson, 2019-10-18:
The pull request was cancelled because of a big lateral overlap at the side of each line. Here is a new plotting function output:
[image]

Seems like it isn't a great idea to change the GGIR inputs around 'visualreport', as if it is changing around a lot with updates, many users may run into problems with it.

Do you think it's best to replace the old plot or add the new plot in the same document?

vincentvanhees, 2019-10-18:
Hi Matt, this looks great. I think it is best to replace the code and fully switch to the new visualisation. 
I cannot think of any reason why a user would want to keep using the old visualisation. If you call the new code also plot5, then it is clear that this is a new version of 'g.plot5'

On a differente note - this plot clearly highlights that the large default windowsizes values for non-wear detection may be too large. Using instead something like windowsizes = c(5, 600, 900) may actually be better. Changing the defaults is tricky, but it may worth introducing a range of default setting updates with the release of GGIR 2.0 (sometime in the upcoming months), which could then also be a good moment to update some of the other default settings and non-intuitive variable names.

m-patterson, 2019-10-18:
Sounds good, I'll put those updates into plot5 and make the appropriate updates in the documentation as well.

m-patterson, 2019-11-14:
Here is the latest update to the visualizations based off recommendations:
[image]
This is what is in the latest pull reqeust

vincentvanhees, 2019-11-21:
Thanks, looks great. Let's go with this.

vincentvanhees, 2019-12-03:
hi Matt, just to let you know that I also had to update onset and waking time extraction, because the most accurate variant of those are stored in acc_onset_ts and acc_wake_ts, not in sleeplog_onset_ts and sleeplog_wake_ts. I tested that it all still works. Thanks again for developing this.

I have merged it in the master branch and will now close this issue.

closed 2019-12-03 by vincentvanhees

m-patterson, 2019-12-03:
apologies for missing that, I think you had in the bug reporting, but I missed it. 

Let me know if you have any feedback on the plotting, I'm happy to make any changes deemed necessary.  

I'll keep a close watch on the group to see if any major bug reports come in over the next little bit for the new plot.