wadpac/GGIR #236: Add step counting to GGIR

m-patterson, 2019-10-07:
I propose using this algorithm: https://ieeexplore.ieee.org/abstract/document/7885036

I propose training / testing be done with this data-set: http://cecas.clemson.edu/~ahoover/pedometer/

JianWang2016, 2020-04-19:
Did I read it correctly that step count is going to release 2.0? That will be fantastic!

One of the outstanding action items for release 2.0
https://github.com/wadpac/GGIR/issues/278

 Discuss with Matt where to store the step detection code and how to include this in the tutorial as one of the extra examples.

vincentvanhees, 2020-04-19:
Not directly. I implemented functionality to facilitate the use of external functions (both R and Python) inside GGIR which are directly applied to the raw calbrated acceleration data. In that way it becomes piloting for other developers to use GGIR as a platform for piloting/implementing their algorithms.

As initial use cases we have:
- The step counting algorithm Matt implemented.
- Kalaivani's sleep detection algorithm from our LRAP project (not published yet as you know).
- Actigraph counts replication with the activityCounts R package.

See draft tutorial: https://github.com/wadpac/GGIR/tree/GGIR2.0/pdf_vignettes_for_github_only

If, at some point, certain external functions become more popular we could consider directly embedding them in GGIR to improve user-friendliness.

I have not worked out a mechanism yet for making these external functions find-able, maybe initially a Wiki page in the GGIR repository with URLs to the various efforts. Although a repository with a back-up of all the code from those external functions could also be valuable.

Re. the step counter it would be good if Matt could decide:
- Whether the algorithm is ready to be shared in public, and if so:
- Whether it should be accompanied by "under development" warnings or maybe with a call for collaborators or funding to validate and improve the algorithm?

m-patterson, 2020-07-15:
Step algorithm to work with GGIR is available here - https://github.com/ShimmerEngineering/Verisense-Toolbox/tree/master/Verisense_step_algorithm

vincentvanhees, 2020-08-09:
I have now added this link to the vignette for External function embedding. So, I think this issue can be closed now.

closed 2020-08-09 by vincentvanhees