wadpac/GGIR #240: Date formatting for actigraph poorly detected.

prinsherbert, 2019-10-11:
I've gotten this error several time, and seen it twice in previous issues.

    Error in g.getstarttime(datafile = datafile, P = P, header = header, mon = mon,  : 
      object 'starttime0' not found

Caused by this [if-branch](https://github.com/cran/GGIR/blob/master/R/g.getstarttime.R#L109), it unfortunately is not transparent. It contains several if-else-if clauses without else-branches for error reporting. Moreover, there is duplicate code. I would assume it's not covering all date formats that Actigraph exports. Please add an else-branch with adequate error reporting, otherwise the code fails with weird errors for proper data.

Other notions:
 * if none of the `as.POSIXlt` succeed, the code fails silently,
 * string-splitting to detect a (regular) expression seems weird, and
 * reading the first line of a file with `read.csv` also seems weird.

Are the latter two common in R?

In Python I would code it like this, I would assume R has some equivalent of this.

    with open(datafile) as f:
      first_line = next(f)

    # Dates in actigraph CSV files can be encoded in different ways. however this format
    # is mentioned in the first line. This lists tuples containing:
    #  - a list of expressions to detect a format in the first line
    #  - a list of expressions to parse a date
    detectors_and_parsers = [
        (
            ["MM[.]dd[.]yyyy", "M[.]d[.]yyyy", "M[.]dd[.]yyyy", "MM[.]d[.]yyyy"],
            ['%m/%d/%Y %H:%M:%S', '%m-%d-%Y %H:%M:%S', '%m.%d.%Y %H:%M:%S']
        ), (
            ["dd[.]MM[.]yyyy", "d[.]M[.]yyyy", "d[.]MM[.]yyyy", "dd[.]M[.]yyyy"],
            ['%d/%m/%Y %H:%M:%S', '%d-%m-%Y %H:%M:%S', '%d.%m.%Y %H:%M:%S']
        ), (
            ["yyyy[.]MM[.]dd", "yyyy[.]M[.]d", "yyyy[.]MM[.]d", "yyyy[.]M[.]dd"],
            ['%Y/%m/%d %H:%M:%S', '%d-%m-%Y %H:%M:%S', '%d.%m.%Y %H:%M:%S']
        ), (
            ["yyyy[.]dd[.]MM", "yyyy[.]d[.]M", "yyyy[.]d[.]MM", "yyyy[.]dd[.]M"],
            ['%Y/%d/%m %H:%M:%S', '%d-%m-%Y %H:%M:%S', '%d.%m.%Y %H:%M:%S']
        )    
    ]

    for detectors, parsers in detectors_and_parsers:
      if any(
        length(unlist(strsplit(topline,"dd[.]M[.]yyyy"))) > 1
        for detector in detectors
      ):
        for parser in parsers:
          parsed_time = as.POSIXlt(starttime,format=parser)
            if is.na(parsed_time) != TRUE:
            break
        else:
          raise ValueError(f"Could not parse ({starttime}) using any format ({', '.join(parser)}) for file: {datafile}")
        break
    else:
      raise ValueError(f"Could understand date format ({firstline}) for file: {datafile}")

    starttime = parsed_time

vincentvanhees, 2019-10-11:
Thanks for highlighting this. I agree that this code section could benefit from some refactoring. I implemented it a long time ago as a quick side project to help a colleague. However, I have hardly worked with Actigraph data myself and also do not have Actigraph device or Actilife software to test any ongoing developments.

So, the best way forward may be:
- A small effort from me to improve both code and documentation inside the code.
- Effort from you (or someone else) to review those changes, propose additional changes with a pull requests, and to test the code on your own files.

prinsherbert, 2019-10-11:
I would have liked to do a PR-request, but the code contains a lot of device-conditional statements and things I do not know, I am afraid to propose anything. I would suggest first to figure out all the date formats supported by Actigraph, one of them is dd-MMM-yy, e.g. 1-Jan-70.

vincentvanhees, 2019-10-18:
I will try to have a look at it, but will be travelling in the upcoming weeks, so probably not before early November. Any change you can tell ActiLife to export the csv file with a different timestamp format as a temporary solution?

prinsherbert, 2019-10-18:
Hi Vincent,

Some of my files had a date format `dd.MMM.yy`, which I fixed by replacing the `MMM` with the right number for that month in each line and also by replacing `dd.MMM.yy` with `dd.M.yyyy` in the header.

This is an ugly solution, especially since the resulting CSV's will contain '19' where '2019' should have been used, which I then post process again.

So for now I am in the clear, you can even drop this issue if you like.

vincentvanhees, 2019-10-18:
Thanks, I will keep the issue open as a reminder to myself.

vincentvanhees, 2019-11-08:
Update: I have now created a PR that should address this issue.
https://github.com/wadpac/GGIR/pull/246

@prinsherbert Would you mind reviewing the PR? I cannot seem to suggest you as a reviewer in the PR, so I have sent you a GitHub invite to become a collaborator of this repo.
Thanks

vincentvanhees, 2019-11-12:
Ignore this, I have merged the PR

closed 2019-11-12 by vincentvanhees

prinsherbert, 2019-11-19:
@vincentvanhees  Thanks for resolving this, my apologies for not responding sooner.

For testing, I have to compile the repository into an R-package and install it, which I have no experience with. Are you still interested in me testing this?

vincentvanhees, 2019-11-21:
I was able to test it on some example files I have,  but you are welcome to double check it.

You can compile and install the GitHub version of an R package from the R command line with the commands I added to #53 a couple of minutes ago: https://github.com/wadpac/GGIR/issues/53#issuecomment-557192715