wadpac/GGIR #257: Axivity data .bin in GGIR

MelissaRS, 2020-01-23:
Hi there,

I am trying to analyse AX3 data using the GGIR package in R. R could not recognize the data as .cvs or .cwa file, so I tried converting the file to .bin. This seemed to work but now in Part 4 I receive the following error:

 1Error in readChar(con, 5L, useBytes = TRUE) : cannot open the connection
In addition: Warning message:
In readChar(con, 5L, useBytes = TRUE) :
  cannot open compressed file '/Users/iCloudMelissa/Documents/Documents/UvA/Bachelorstage/Oefenresults//output_Oefendata/meta/ms3.out/NA', probable reason 'No such file or directory'

GGIR does create files in the output directory: meta with basic, ms2.out, ms3.out ms4.out and ms5.out and a results folder with QC folder and session info files. I do not get any other output than this. Can someone please provide tips/tricks for solving this?

vincentvanhees, 2020-01-23:
GGIR is able to read AX3 files in cwa wav and csv directly. There is no need to convert to bin format.

1. Does the AX3 cwa file load in OMGUI software?
2. What is the duration of the recording? GGIR is designed for multi-day recording and not for short recordings of less than a few hours.
3. Did you try reading the a fraction of the file directly with GGIR function g.cwaread?

Please share any other info you may have about why the cwa file does not read, the error messages you encounter, and the exact commands you have used to try to load the data.

MelissaRS, 2020-01-23:
If I put the .cwa file in the input data directory I receive an error stating "No files to analyse. Check that there are accelerometer files in the directory specified with argument datadir". Once I add a .bin file the error disappears and the GGIR starts running.

1. The AX3 file does load in the OMGUI software, but not om my computer (I am using Mac and the OMGUI software is only for Windows.
2. The duration of the recording is 7 days I believe. I am using the example data from the website: https://axivity.com/downloads/ax3. I directly downloaded this .cwa file and tried to analyse it using GGIR.
3. I tried using this function but I receive output saying Error in file.info(fileName) : object 'longitudinal_data' not found.

The exact code that I use to run the GGIR is:
g.shell.GGIR(mode=c(1,2,3,4,5),datadir="~/Documents/Documents/UvA/Bachelorstage/Oefendata/",outputdir="~/Documents/Documents/UvA/Bachelorstage/Oefenresults/",do.report=c(2,4,5),strategy = 1, hrs.del.start = 0,hrs.del.end = 0, maxdur = 9, includedaycrit = 16, qwindow=c(0,24), mvpathreshold =c(100), bout.metric = 4, excludefirstlast = FALSE, includenightcrit = 16, def.noc.sleep = 1,outliers.only = TRUE, criterror = 4,do.visual = TRUE, threshold.lig = c(30),threshold.mod = c(100),threshold.vig = c(400),boutcriter = 0.8,boutcriter.in = 0.9,boutcriter.lig = 0.8, boutcriter.mvpa = 0.8, boutdur.in = c(1,10,30), boutdur.lig = c(1,10),boutdur.mvpa = c(1), timewindow = c("WW"), visualreport=TRUE)

I have tried changing/adding several arguments to the code but this does not make any difference.

vincentvanhees, 2020-01-23:
GGIR is written and tested for Linux, OSx and Windows.

Did you install GGIR from source or from CRAN?

If you are sourcing the R functions from a cloned copy of the repo then please note that to read cwa files you will need to compile the c++ code in the src folder of GGIR as in:
library("Rcpp")
sourceCpp(paste0(pathtoGGIRcode,"/src/numUnpack.cpp"))
sourceCpp(paste0(pathtoGGIRcode,"/src/resample.cpp"))

Installing GGIR directly from CRAN will avoid this step, because your OSx will do the compilation for you.

Are you able to read the AX3 wav and csv exports from the cwa file? The wav format does not depend on the c++ code and the csv data relies on it to a lesser extent.

vincentvanhees, 2020-01-23:
Another tip: use argument ` print.filename=TRUE,` in g.shell.GGIR to tell GGIR to print the filename that it is trying to read to the console. This helps to check that you are trying to read an actual accelerometer file. Useful if you forget to remove non-accelerometer files from your data folder.

vincentvanhees, 2020-01-23:
I am able to process that exact file in Ubuntu with the current GGIR version (1.11-0).

MelissaRS, 2020-01-24:
I did install the GGIR from CRAN. 

I cannot open the AX3 .cwa file itself because I have no programms that can open .cwa. I tried converting the .cwa into .csv but this results in a file of 60 million rows in excel. Excel can open this but it takes a very long time to load.

I used the argument `print.filename=TRUE` but I still receive the following error:
Error in g.part1(datadir = datadir, outputdir = outputdir, f0 = f0, f1 = f1,  : 
No files to analyse. Check that there are accelerometer filesin the directory specified with argument datadir.

The only file in the directory is the .cwa data, but it appears that GGIR does not recognize this file. Once I add a .bin file, GGIR does try to analyse the .bin file but I receive the error in my first comment.

closed 2020-01-24 by MelissaRS

vincentvanhees, 2020-01-24:
Sorry to hear the cwa reading doesn't work. GGIR has a unit test for reading tiny dummy cwa file, which passes on OSx. So, I am not sure where it goes wrong for you.

As an alternative to the bin-route you have chose, you may want to check out the [command line tools](https://github.com/digitalinteraction/openmovement/blob/01362d5155002d0bf326ae8825c33b953da2b7c5/Software/AX3/cwa-convert/c/README.md) the AX3 developers put online to convert cwa to csv.

I conclude from your filepath that you are Dutch. Do not hesitate to pm to discuss further. Would be good to know what local researchers are doing in this area. I am frequently at VU and at Amsterdam Science park. Cheers, Vincent

MelissaRS, 2020-01-24:
Mijn probleem is inmiddels opgelost! Heel vervelend, ik heb de data van AX3 verwijderd en opnieuw gedownload. Vervolgens heb ik deze data met exact dezelfde naam in dezelfde directory gezet en toen herkende GGIR de .cwa file ineens wel. Toch enorm bedankt voor je hulp!

Ik ben overigens inderdaad Nederlands. Ik ben bezig met een bachelorstage van de UvA (Science Park) en hou me hiervoor bezig met wat accelerometer data. Nogmaals bedankt voor het mee denken! Groet, Melissa