wadpac/GGIR #277: Create unit tests that compare results with the same file

m-patterson, 2020-04-15:
Generate a 24h file that has known results for key variables in the unit testing.

Compare key results to what is expected to make sure that any new changes to code aren't changing any results. 

Details: Instead of creating a data file of random numbers, create a data file that we know is the same every time.

vincentvanhees, 2020-04-15:
Great idea, but keep in mind that our current unit tests already sets the seeds constant by which the test file is always identical.The file is artificial but triggers most of the GGIR functionality. The file is only 3 Hertz, the minimum required for GGIR to run. I took this approach because I wanted to minimize the duration of the testing and because CRAN wants the package to stay as compact as possible.

So, maybe consider:
- Complement the existing tests with tests focused around a real part 1 milestone data file rather than a raw data file.
- Create a separate GGIR package named GGIRdata with example milestone data and tiny testfiles (bin, cwa, csv, wav) that are currently in inst.testfiles.

Test coverage is tracked here: https://codecov.io/gh/wadpac/GGIR. Once in a while I go there to review which functions are most in need for better test coverage.

vincentvanhees, 2020-12-04:
I am close this issue, feel free to re-open if you think this still needs work.

closed 2020-12-04 by vincentvanhees