wadpac/GGIR #282: Warnings in plot5

vincentvanhees, 2020-04-17:
See google group discussion [here](https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/rpackageggir/PEUnOeEvIjk/dHohGDm8CAAJ)

Performing a test in a IF, ELSE of ELSE IF statement should always produce a single Boolean result.  If wake_loc is longer than 1 and tested for == 0 it will produce a vector of results. So, either avoid that wake_loc can get longer than 1 or change code such that it produces a single result for every IF, ELSE and ELSE IF statement.

m-patterson, 2020-04-17:
hey Vincent, I made these changes, along with changing the default nnights value to c().

However, I made a mistake by pushing my branch to the GGIR master, instead of my forked version and then doing a pull request. Sorry about that.  Let me know if you want to roll it back and for me to make a pull request. I ran the testing on my machine and it worked fine.

Thanks, Matt

vincentvanhees, 2020-04-19:
Thanks Matt for doing this so quickly.
No problem about the push to Master branch, these are minor changes to a function nobody else is working on.

closed 2020-04-19 by vincentvanhees