wadpac/GGIR #287: Add images to sleep functionality explanation

m-patterson, 2020-04-24:
Vignette: add some images to aid in the sleep functionality explanation

m-patterson, 2020-04-27:
Here is the updated summary of how the sleep detection works:

1. Get sustained inactivity Bouts (SIBs) on all data.
2. Define Sleep Period Time (SPT) window as the start of the first SIB and the end of the last SIB that overlap with a guiding method as detailed below.
3. Consider SIB during SPT window as sleep time.
4. Consider non-SIB during SPT window as nocturnal wake time.

vincentvanhees, 2020-06-04:
[CRAN results](https://cran.r-project.org/web/checks/check_results_GGIR.html) indicate that our documentation is getting too big when compiled for windows. Probably we will be asked to address this before as a condition for the next GGIR release on CRAN. 

It would be good to know whether moving some of the documentation to separate pdf vignettes could address this. Another approach could be to avoid including images in the vignette and refer to external image sources or moving entire parts of the documentation to external locations. I was hoping that we could avoid that documentation gets scattered across many places though.

Maybe good to investigate how other R packages have solved this.

vincentvanhees, 2020-08-09:
I now see on stackoverflow [link](https://stackoverflow.com/questions/53819970/r-package-libs-directory-too-large-after-compilation-to-submit-on-cran) and also elsewhere that the note about large folder size can be ignored. So, I think for the time being we can continue with the vignette as it is and just be considerate with file size when adding images.

Putting it on my website is not ideal, because that would make it less accessible for collaborative contributions. A better idea may be to create a [gitbook](https://www.gitbook.com/). This will then live in a separate GitHub repository and is designed for these kind of things. To do this it may be good to start by defining a structure and outstanding tasks, in order to make it easy for many people to contribute.

closed 2021-12-09 by vincentvanhees