wadpac/GGIR #289: Enable unzip, process and zip of gunzipped csv files

vincentvanhees, 2020-04-26:
Initially faciltiate this for Linux

vincentvanhees, 2020-04-26:
I have now implemented a solution in a new development branch.
It turned out that fread from the read.table package can handle gz-files if R.utils is installed. So, I have added R.utils to the dependency list.
However, we also use read.csv a couple of times. For now I have added a small new function 'checkgz' that helps read.csv to in handling gz-files.

Another solution would be to replace all read.csv calls by fread calls, but then there is a risk for functional consistency, which would have to be checked.

vincentvanhees, 2020-04-26:
Outstanding action points:
- [ ] Add unit test
- [ ] Update documentation
- [ ] Explore whether this can easily be extend to other types of zip-formats

muschellij2, 2020-05-04:
The base `read.csv` already can read `csv.gz` files.  Also, `readr::read_csv` can do that quicker.

vincentvanhees, 2020-05-21:
This is now addressed in the branch: issue289_unzipping. I will make a pull request in the upcoming weeks.

closed 2020-05-21 by vincentvanhees