wadpac/GGIR #292: Streamlining release process

vincentvanhees, 2020-05-01:
Hi @jhmigueles and @m-patterson. To streamline the GGIR release process I would like to propose the following guideline for all contributors.

Let me know what you think. When you are happy I will add this to the Contributor instructions in the README file:

### 1. Announce new features/ bug fixes

Always use GitHub issues to announce that you are working on a new feature or bug fix like we do at the moment. In this way we can provide feedback to each other at an early stage. Informing me in an e-mail that you are working on something is not enough.

### 2. Pull request for a new feature / bug fix

1. Make sure your branch is up to date with the master branch as in `git checkout master, git pull, git checkout yourbranch, git merge master`.
2. Make sure the change log `inst/NEWS.Rd` is up to date and that it says "GitHub-only-release date" rather than "release date"
3. Make sure the **third** (last) digit in the version number is incremented by one relative to the master branch and the date is the present date. This applies to the files DESCRIPTION, Citation.cff (not the cff-version, but the version on line 56), GGIR-package.Rd and NEWS.Rd file. Use function prepareNewRelease.R in the root of GGIR to double check that version number and date are consistent between these files.
4. Update package contributor list if new people have contributed. 
5. Run `R CMD check --as-cran` to make sure all tests and checks pass.
6. Make a pull request and invite at least one of us (Jairo, Matt, Vincent) to review it.
7. If the reviewer is happy, wait till the end of the month and one of us will merge all open pull requests and create new a GitHub release. 

### 3. GitHub releases

1. When you do this please inform the others that you are going to do this. 
2. Re-do steps 2.1-2.5.
3. Create new GitHub release. Note that GitHub releases require a release name, I typically choose a random name of a city or town in South America. Whatever you choose this should be an easy to read and remember.

### 4. CRAN releases

1. Create GitHub issue at least 4 weeks before the intended CRAN release announcing the release and indicating what will be in the release and a to do list.
2. A CRAN release should not come with major changes that have not been covered by any of the GitHub releases, except from full version updates (1.0 -> 2.0 -> 3.0 -> ...). If a full version update is the case, create a branch in the wadpac/GGIR repository dedicated to the new release. 
3. When everything looks ready for the release, repeat the same process as for the GitHub release with a few differences: 
    - In the change log it should now say "release data" rather than "GitHub-only-release date".
    - Second digit in the version number is incremented by 1 relative to the current CRAN version.
    - Check whether a new R version has been released or is coming up and make sure GGIR is also tested with that version.
    - Run `R CMD check --as-cran` also with `r-devel`, which is a more thorough check of the package.
    - All of us will review the new release, and when all are happy I will submit it to CRAN as this needs to come from my e-mail address.

jhmigueles, 2020-05-01:
Hi @vincentvanhees . Everything looks great from my side. Nothing to add.

m-patterson, 2020-05-01:
That looks like a really good / thorough updating protocol to me, I have nothing to add

vincentvanhees, 2021-07-16:
Closing this issue, because release process is now clarified in https://github.com/wadpac/GGIR/blob/master/RELEASE_CYCLE.md

closed 2021-07-16 by vincentvanhees