wadpac/GGIR #299: plot5 not working when viewingwindow=0

m-patterson, 2020-05-19:
Two updates to make to plot5 functionality

- Argument viewingwindow does not work when set to 0 (plots centered at midnight rather than noon), the code only runs when this is set to 1 (plots centered around noon). This issue may have originated in the autumn, but we failed to notice it. Do you think you can get it to work again?

- When the reports are generated without access to the original data folder (datadir) and none of the parts are re-run. GGIR does not know what value to assign for f1 and sets it to zero. To address this, can you please add the following line after line 12:
if (f1 == 0) f1 = N_milestone_data_p4

vincentvanhees, 2020-06-04:
> When the reports are generated without access to the original data folder (datadir) and none of the parts are re-run. GGIR does not know what value to assign for f1 and sets it to zero. To address this, can you please add the following line after line 12:
> if (f1 == 0) f1 = N_milestone_data_p4

I just addressed the aspect as part of final edits to prepare Github-only release 2.0-1.

vincentvanhees, 2020-06-18:
Thanks Matt, I have merged the PR and will now close this issue. Later today I will create a new GitHub release from this and a few other minor updates.

closed 2020-06-18 by vincentvanhees