wadpac/GGIR #300: Error in `[<-`(`*tmp*`, i, (cnt + 19), value = sd(nightsummary.tmp$sleeponset[indexUdef],  :    subscript out of bounds

lydiantoinette, 2020-05-22:
Hello, 

I'm wondering if anyone has seen this error: 
Error in `[<-`(`*tmp*`, i, (cnt + 19), value = sd(nightsummary.tmp$sleeponset[indexUdef],  : 
  subscript out of bounds

It comes up after Report part 4.

Thank you,
Lydia

Here is the shell script that I am using:
g.shell.GGIR(#-------------------------------
             # General parameters
             #-------------------------------
             mode=mode,
             datadir=datadir,
             outputdir=outputdir,
             studyname=studyname,
             f0=f0,
             f1=f1,
             overwrite = TRUE,
             do.imp=TRUE,
             idloc=1,
             print.filename=FALSE,
             storefolderstructure = FALSE,
             #-------------------------------
             
             # Part 1 parameters:
             #-------------------------------
             windowsizes = c(5,900,3600),
             do.cal=TRUE,
             do.enmo = TRUE,
             do.anglez=TRUE,
             chunksize=1,
             printsummary=TRUE,
             desiredtz= "America/Toronto",
             #-------------------------------
             # Part 2 parameters:
             #-------------------------------
             strategy = 1,
             ndayswindow=7,
             hrs.del.start = 1,
             hrs.del.end = 1,
             maxdur = 35,
             includedaycrit = 16,
             L5M5window = c(0,24),
             M5L5res = 10,
             winhr = c(5,10),
             qlevels = c(c(1380/1440),c(1410/1440)),
             qwindow=c(0,24),
             ilevels = c(seq(0,400,by=50),8000),
             mvpathreshold =c(100,120),
             #-------------------------------
             # Part 3 parameters:
             #-------------------------------
             timethreshold= c(5,10),
             anglethreshold=5,
             ignorenonwear = TRUE,
             #-------------------------------
             # Part 4 parameters:
             #-------------------------------
             excludefirstlast = FALSE,
             includenightcrit = 16,
             def.noc.sleep = c(),
             loglocation= c(),
             outliers.only = FALSE,
             criterror = 4,
             relyonsleeplog = FALSE,
             sleeplogidnum = TRUE,
             colid=1,
             coln1=2,
             do.visual = TRUE,
             nnights = 35,
             #-------------------------------
             # Part 5 parameters:
             #-------------------------------
             # Key functions: Merging physical activity with sleep analyses
             threshold.lig = c(40),
             threshold.mod = c(100),
             threshold.vig = c(400),
             excludefirstlast = FALSE,
             boutcriter = 0.8,
             boutcriter.in = 0.9,
             boutcriter.lig = 0.8,
             boutcriter.mvpa = 0.8,
             boutdur.in = c(10,20,30),
             boutdur.lig = c(1,5,10),
             boutdur.mvpa = c(1,5,10),
             timewindow = c("WW"),
             #-----------------------------------
             # Report generation
             #-------------------------------
             do.report=c(2,4,5),
             visualreport=TRUE, dofirstpage=TRUE,
             viewingwindow = 1)
## End(Not run)

vincentvanhees, 2020-06-04:
Can you please include output from command sessionInfo().
Does this happen with a specific recording or with all your recordings?

lydiantoinette, 2020-06-09:
Hi Vincent,
This seems to be happening for multiple recordings unfortunately (different lengths too). Below is the output from > sessionInfo():

R version 3.6.1 (2019-07-05)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows >= 8 x64 (build 9200)

Matrix products: default

locale:
[1] LC_COLLATE=English_United States.1252  LC_CTYPE=English_United States.1252   
[3] LC_MONETARY=English_United States.1252 LC_NUMERIC=C                          
[5] LC_TIME=C                             

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] GGIR_2.0-0

loaded via a namespace (and not attached):
 [1] compiler_3.6.1    parallel_3.6.1    tools_3.6.1       yaml_2.2.1        Rcpp_1.0.2        codetools_0.2-16 
 [7] doParallel_1.0.15 iterators_1.0.12  foreach_1.5.0     data.table_1.12.8

Thanks,
Lydia

vincentvanhees, 2020-06-16:
You are using 2.0-0. Can you update to GGIR 2.0-1 from GitHub.
```
library(devtools)
install_github("wadpac/GGIR")
```
and try again? You may need to restart R.

I am not sure whether this will solve it, but it would be good to know whether the issue also exists in the latest GGIR version.

lydiantoinette, 2020-06-16:
I have just tried this on 2.0-1, and am now getting a slightly different error (same script as before):

Error in `[<-`(`*tmp*`, i, (cnt + 13), value = mean(nightsummary.tmp$duration_sib_wakinghours_atleast15min[indexUdef],  : 
  subscript out of bounds

vincentvanhees, 2020-06-18:
I updated the GitHub version of GGIR this afternoon to version 2.0-2. Can you try again?
If that does not address it, do you see any scope to pay for my time to look more into it based on an example data? €150. I would like to avoid the impression that all GGIR problems are my problems, it is free software after all. Happy to help now and then with simple issues, but it looks like this is requiring more time investment. Please send me an e-mail if you are interested (v.vanhees@movementdata.nl).

vincentvanhees, 2020-06-28:
This should be fixed now here on GitHub. Please re-install GGIR with
```
library(devtools)
install_github("wadpac/GGIR").
```
Version number is still 2.0-2 as this was a very small edit.

closed 2020-07-02 by vincentvanhees