wadpac/GGIR #305: full example with data?

aaronjfisher, 2020-06-15:
Hello!

I'm trying to get acquainted with GGIR, and might hope to contribute sometime soon, but for now I'm just trying to see how it all fits together. Is there a full, run-able example that I can download and try? All of the examples seem to be "not run" examples. 

If there isn't one, could you add one? I'm not sure, but I think a streamlined way to do this might be to add a GGIR::generate_data function, save that data using the base::tempdir function, and then use this temporary directory as the data path in subsequent steps. Would that work?

As an aside, the thing I'm specifically trying to understand about the package right now is why g.getM5L5 isn't called inside of g.part5. Rather, g.part5 seems to have additional code for M5L5. Wouldn't it be better to consolidate and have all the code in one place? Again, the reason I'm trying to load up an example is so that I can try to parse out what's different about these two code chunks. I don't want to raise that as a separate issue without understanding it better, but if you have time to comment on that too, that would be awesome!

Thanks very much,
-Aaron

vincentvanhees, 2020-06-16:
Hi Aaron,
GGIR does not have a full data example because raw data files are too large to include. GGIR has functions (create_test_...) that generate simulated test data to aid the unit tests, but I would not recommend fully relying on them to learn about GGIR. They are designed to generate just enough artificial data to trigger all the functions.

The best way to explore GGIR is with real data. Axivity put an example file on their website, which you may want to use: http://www.axivity.com/files/resources/longitudinal_data.zip. Further, you might want to ask researchers you work with to give you an example file from GENEActiv, Actigraph, or any of the other compatible accelerometer brands.

Thanks for pointing out the potential overlap in functionality. I fully agree that it would be good to avoid redundancy. I will have a look whether I can address it.
Vincent

aaronjfisher, 2020-06-22:
Perfect, thanks Vincent!

closed 2020-06-22 by aaronjfisher

aaronjfisher, 2020-06-26:
If I'm not overstepping, I'm going to post the code overlap as a separate (minor) issue.