wadpac/GGIR #314: read.myacc.csv vignetter

jhmigueles, 2020-07-05:
I am working on a tutorial vignette for the read.myacc.csv function.

m-patterson, 2020-08-12:
@jhmigueles any updates on this? Even if you have a preliminary version, would it be possible for me to have a look at it? Trying to get it working.

jhmigueles, 2020-08-12:
I prepared a (very) preliminary version in the [issue314_readmyacccsv_vignette](https://github.com/wadpac/GGIR/tree/issue314_readmyacccsv_vignette) branch. I plan to work more on this on September, any input is welcome.

m-patterson, 2020-08-12:
Thanks, I'll have a look in there

vincentvanhees, 2022-04-06:
@jhmigueles given that you are planning to continue working on this, here some additional info:

You may want to be aware that I tried to cover all main read.myacc.csv use-cases in [this](https://github.com/wadpac/GGIR/blob/master/tests/testthat/test_read.myacc.csv.R) unit test.

Please find below code snippets I found in my code history relating to reading ad-hoc .csv data from various accelerometer formats. I cannot guarantee that they still work as these accelerometer brands may have changed their formatting. I propose not adding them to the vignette because that creates pressure on us to always make sure that these examples are correct for each brand.  Once a user decides to work with ad-hoc .csv format it is their responsibility to figure out what the input arguments should be. I think the main purpose of the vignette should be to demonstrate the process of finding the right input arguments bases on some hypothetical examples similar to the unit test.

### Verily (when configured to export raw data for research)

```
rmc.dec=".",
rmc.firstrow.acc = 2,
rmc.col.acc = c(2:4),
rmc.col.time=1,
# rmc.col.wear=5,
rmc.unit.acc="g",
rmc.unit.time="POSIX",
rmc.format.time = "%Y-%m-%d %H:%M:%OS",
rmc.desiredtz = "America/Los_Angeles",
rmc.sf = 30,
rmc.skip=0, 
rmc.noise = 0.02,
```

### Empatica E4

 ```
rmc.dec=".",
rmc.firstrow.acc = 20,
rmc.firstrow.header = 1,
rmc.col.acc = c(2,3,4),
rmc.col.temp=5,
rmc.col.time=1,
rmc.unit.acc="bit",
rmc.unit.temp = "C",
rmc.unit.time="UNIXsec",
rmc.bitrate = 8,
rmc.dynamic_range = 2,
rmc.headername.sf="acc_sample_rate",
rmc.headername.sn="device_serial_number",
rmc.headername.recordingid="recording_ID",
rmc.header.structure = c(),
rmc.check4timegaps=TRUE,
rmc.header.length=19,
# rmc.skip=0,
rmc.origin = "1970-01-01",
rmc.desiredtz="America/New_York",
rmc.unsignedbit=FALSE,
```

### Activpal .csv file

```
rmc.nrow = Inf,
rmc.dec = '.',
rmc.col.acc = 2:4,
rmc.col.time = 1,
rmc.unit.acc = 'g',
rmc.unit.time = "ActivPAL",
rmc.dynamic_range = 2,
rmc.desiredtz = "Europe/London",
rmc.sf = 20,
rmc.origin = "1970-01-01",
```

jhmigueles, 2022-04-19:
@vincentvanhees you may want to look at the read.myacc.csv vignette in the [issue314_readmyacccsv_vignette] branch. Let me know your thoughts. I am going to keep working on the documentation in this branch as we planned.

vincentvanhees, 2022-05-05:
Closing, because has been fixed in PR #549

closed 2022-05-05 by vincentvanhees