wadpac/GGIR #317: A particular csv-file is not recognized

Weyaaron, 2020-07-10:
I'm runnnig version GGIR 1.11-0 on Arch Linux and R-Version 4.0.2, with data from
Actilive-Devices. 

The filename is:

>  "MOS2E31191189 (2020-03-26)RAW.csv"

The Header of the file in question is:

> 
> ------------ Data File Created By ActiGraph GT3X+ ActiLife v6.13.4 Firmware v1.9.2 date format dd.MM.yyyy at 100 Hz  Filter Normal -----------
> Serial Number: MOS2E31191189
> Start Time 10:15:00
> Start Date 26.03.2020
> Epoch Period (hh:mm:ss) 00:00:00
> Download Time 10:41:48
> Download Date 09.04.2020
> Current Memory Address: 0
> Current Battery Voltage: 3.85     Mode = 12
> --------------------------------------------------
> Accelerometer X,Accelerometer Y,Accelerometer Z
> [...]

This file is not recognized by the package, but should be. 

This is one my my first issues on github, if you need additional data let me
know.

vincentvanhees, 2020-07-16:
Header looks good to me. Please post error message. Also please try running latest version of GGIR.
Thanks

Weyaaron, 2020-07-21:
I ran the latest version of ggir, this has fixed this issue.

closed 2020-07-21 by Weyaaron