wadpac/GGIR #318: HDCZA sleep detection error on first night

m-patterson, 2020-07-15:
The HDCZA algorithm to detect sleeponset and wake times (without sleep logs) seems to sometimes throw up an error on the analysis of data from the first night. 

Here is an example of some poor behaviour:
[image]

However, when we use the same data, but start from noon on the first day, the results improve dramatically:
[image]

This leads me to think that there may be a bug inside the HDCZA algorithm that prevents it from detecting properly in some cases. 

Goal - find the bug and fix it.

Issue - do not create a fix that will cause errors on other tests (test on as much data as possible)

m-patterson, 2020-07-15:
The error doesn't seem to be in the HDCZA algorithm, but in how the detected points are handled after the algorithm.

This is the relevant point in g.sid.det:
[image]

The code block transfers the ws3 sample on which sleeponset and wakeup were determined into hours. On the first pass through (when j == 1), the code assumes that the start of the recording is equivalent to the start time of the data sent into the HDCZA algorithm (which is noon - noon). So, if the recording starts before noon, an erroneous time will be generated for sptwindow_HDCZA_start and sptwindow_HDCZA_end. 

This should be updated to do the startTimeRecord only if there is less than a full noon-noon chunk passed into the HDCZA algorithm.  Perhaps, a good test for this is to see if (qqq1 == 1), as qqq1 is the start index in the 5sec epoch data for the chunk that is passed to the HDCZA algorithm. I'll perform some tests to check this.

vincentvanhees, 2020-07-15:
Thanks for investigating. I am travelling at the moment, but back next week. Your explanation sounds plausible. 
I did most of the testing in a data set where the first and last night always had to be ignored. So, testing of detection in first and last night may not have been as rigorous as for the other recording nights.

vincentvanhees, 2020-07-16:
Thanks, unit test issue now fixed, and package is building again. I will close this issue now.

closed 2020-07-16 by vincentvanhees