wadpac/GGIR #321: duplicated values in part4 csv output

m-patterson, 2020-07-21:
With the latest master version, I am getting duplicated nights in part4 csv output, like this:

[image]

Because of this, the nightsummary2 dataframe has values on repeated nights and that is what gets written to the csv file.

When, I looked into the problem, it seems the code in g.report.part4.R is not working correctly. In this code, it opens the corresponding ms5 file to get the non-wear percentage per night. This output can have multiple values for one night and the comments in the code state: '# some days in part 5 can have two SPTs, remove first and keep second SPT'. A problem occurred in one of my test files because there were three nonwear_perc_spt elements for one night:

[image]

I'm not too sure if it is best to try to fix this higher up the chain (ie, where ms5 is being saved) or best to fix it here. For now, I'll work on a development branch to fix it here, but let me know if it should be fixed higher up the chain. The fix I will work on will only keep the final element of any repeated values and ignore the preceding elements.

I'll be sure to work on this in a dev branch and submit a pull request.

m-patterson, 2020-07-21:
With the fix in the dev branch here is outputp5 before searching for duplicates:
[image]

Here is outputp5 after dropping initial duplicates:
[image]

The resulting part4 csv file has no repeated values in it:
[image]

However, it seems a bit surprising that it is keeping the first nights sleep, when it is all nonwear and sleeponset and wakeup were really close to one another in the morning:

[image]

vincentvanhees, 2020-07-21:
I think I fixed the double night issue in May with [this](https://github.com/wadpac/GGIR/commit/2c3805a6d6f3b246f2568350efeae07bafd4c31b#diff-ee56763e039c55f2cd5f74b37357969a) commit: 

Comparing that with your commits from today [(link)](https://github.com/wadpac/GGIR/commit/2c3805a6d6f3b246f2568350efeae07bafd4c31b#diff-ee56763e039c55f2cd5f74b37357969a) indicates that your branch is not in sync with my commit.

vincentvanhees, 2020-07-21:
... sorry, it is in sync, but possibly my fix in May does not address all possible use cases.
What I do in part 4 report is to extract from specific variable from the part 5 results and merge it into the part 4 results, because it would be too much work to also calculate that variable in part 4. In the process of merging something went wrong causing double nights. However, I thought I fixed this. Could you send me an example milestone file to help investigate this?

vincentvanhees, 2020-07-22:
I have now used GGIR form your development branch to process the test files you sent.

In my output it correctly removes the first night. However, while looking into this I spotted and fixed a different issue. For your test file the ID number in part 5 is stored as "1" while in part 4 it stored as "01", which complicates the merging of the variable. I have now expanded your branch to resolve the issue when it happens.

The output I then get is:
[image]

I am not sure why the first night was not removed for you. Maybe you could delete the csv report to be sure it is replaced?

The multiple windows per calendar date are plausible in part 5, but in part 5 I decided to make a pragmatic decision to only load one night. Maybe taking the average would be better. On the other hand it is not a major issue as double SPT windows rarely occur in GGIR. You would really need someone who stays up for 24 hours. I have left a note inside the code to remind myself in case we work on this again.

m-patterson, 2020-07-23:
Hi Vincent,

I copied a version of the issue315_doubleresults branch and ran it on the test in question. It solved the problem of double results, however, it still included night 1 in the 'part4_nightsummary_sleep_cleaned.csv' file. I pushed a version on the branch that I used to my remote, just in case that is of any use. Here are the part4 results:

[image]

I wonder if it due to the fact that I have 'excludefirstlast = FALSE' in my entry function?

vincentvanhees, 2020-07-24:
Hi Matt, I used the milestone data you provided and worked on the code with mode=c(), so in that way your processing decisions should have been preserved. Can you also send me a config file such that I can reproduce the error.

closed 2020-07-27 by m-patterson