wadpac/GGIR #322: Detected non-wear in the output of part 2 does not reflect reality

prinsherbert, 2020-07-24:
While processing Acrigraph CSV-files of several days with different non-wear moments or 'night time' between these days, the output of step 2 creates an ENMO signal that has the same non-wear period for each day.

Moreover, the ENMO has a signal at place where the original accelerations were all zero. I have found this very strange. How can the ENMO be non-zero whereas the raw signal was zero?

Here's a minimal example on synthetic data.

[Google Colab minimal example](https://colab.research.google.com/drive/1IePgGyTCYZeBrWNXRLZEFAgNdE38ZIjc?usp=sharing)

This plot shows the result, the red curve is only high when the original signal is active. It's clearly visible that it has different night times. The blue curve is only high when the ENMO of step 2 is non-zero. I can't see why the ENMO should be non-zero when the original signal is flat.

[image]

vincentvanhees, 2020-07-24:
GGIR imputes the time segments that it classifies as non-wear by the average values of similar time points on different days of the recording. There is only one day with valid data for hours 0-2, so these hours are then used as imputation for all other days. 

If you are concerned that this produces unrealistic time series, you can turn off the imputation with argument do.imp=FALSE. 
If the study protocol is to wear the sensor for 24/7 then this scenario is very unlikely in real life, but it is more common in situations where the study protocol requires the person to take off the senor during specific parts of the day. In that case setting do.imp=FALSE may be the best solution. I have not encountered this in many projects, so haven't really thought about a solution yet.

prinsherbert, 2020-07-28:
Our experiment deals with participants that are allowed to and will typically go to sleep without the sensor, and they will choose their bed and wake-up times. Hence they differ from day to day.

However, I've noticed the imputation behavior, which makes the results appear if participants wake up and go to bed at the exact same times on different days, didn't occur for all our participants. Is there a reason the imputation isn't done for every day, even though `do.imp` was set?

vincentvanhees, 2020-08-28:
Imputation is only done for time segments where there is valid data on the same timepoint on different days of the recording.

See also variable complete_24hcycle in the part2 output. This will be less than 1 if accelerometer is systematically not worn at certain times of the day across recording days.

vincentvanhees, 2020-09-25:
I am closing this issue now, but feel free to re-open if you feel this needs further clarification.

closed 2020-09-25 by vincentvanhees