wadpac/GGIR #324: Bug converting to time from daysleeper option in HDCZA sleep detection

m-patterson, 2020-07-27:
I have a test in which the HDCZA algorithm detects sleep close to noon, so the algorithm reprocess the sleep period using data from 6pm to 6pm (instead of noon to noon). However, when the code translates these sleep/wake detection points, it does not take into account that the times were starting from 6pm instead of 12-noon. 

After the detection (HDCZA) algorithm is run, the code goes into this section to translate the sample number from the chunk sent into HDCZA into times:
[image]

There is no option for starting at 6pm instead of 12-noon. 

I will add this in a new feature branch and make a pull request.

m-patterson, 2020-07-27:
Before the fix, this test resulted in ~2pm sleeponset time and ~3am wake time:
[image]

However, upon inspection it did go into the daysleeper aspect of HDCZA detection and returned samples 1 and 10275:
[image]

This corresponds to hrs 12 and 2.2, which make sense considering they will then be used as 'guiders' to get the final sleeponset and wakeup times. How can we have a sleeponset guider starting at 12-noon, when the chunk of data started at 6pm? The code is not taking into account the adjusted start time.

With this fix in the code (add daysleep_offset) to the code:
[image]

We still have 1 and 10275 as the detection points from the HDCZA:
[image]

However, the resulting times are correct, considering the 6pm start of the block:
[image]

I also cleaned up the legacy code in there.

vincentvanhees, 2020-08-09:
thanks, now merged into master

PR #325 (2020-08-09): fix bug in daysleeper detection
Fixes #324 

Small bug fix to ensure that correct start block time is used when performing daysleeper detection

Cleaned up some of the legacy code that was commented out as well.

@vincentvanhees - not sure if you want this fix into v2.1-0 or not, I put it here for now, but let me know if you'd rather hold off and save it for the next release. Is a relatively small fix, so hopefully ok for this release.
closed 2020-08-09 by vincentvanhees