wadpac/GGIR #326: Numerically unstable moving average

prinsherbert, 2020-07-28:
Cumulatively adding large numbers of floating point numbers is numerically unstable, since the precision will decrease as the numbers gets larger.

Consider changing [this](https://github.com/wadpac/GGIR/blob/master/R/g.applymetrics.R#L31) to apply `cumsum(diff( ... ))` instead of `diff(cumsum( ... ))`.

In Python this would be something like:

    def moving average(x, width):
        x = [0] + x # insert a zero at the start
        
        # start from the average of the first `width-1` numbers
        # then using cumsum, x[0], x[1], ... and x[width-1], x[width], ... will be respectively subtracted and added for each new element
        return x[:width].sum() / width + cumsum(x_[width:] - x_[:-width]) / width

[More information](https://stackoverflow.com/questions/13728392/moving-average-or-running-mean/60673218#60673218)

vincentvanhees, 2020-08-07:
Thanks for pointing this out.
Reverting diff(cumsum( to cumsum(diff is not that easy, because the order diff(cumsum is critical for derving mean.

The Python/Numpy functions in your link use a rolling sum, which does not exist in base R. Possibly there are solutions in other R packages, but I am hesitant with adding more dependencies to GGIR.

The data that is being aggregated involves values equal to or less than 90 (in case of angle), which makes that even a year worth of 100 Hertz times the value 90 will not cumsum to more than 1e+11 (100 * 3600 * 24* 7*52*90), which is below typical machine precision. So, I think the current implementation is sufficient for the data it is designed to work with.

I agree though that it is a potential problem if higher floating point values would be used.

closed 2020-08-07 by vincentvanhees

prinsherbert, 2020-08-14:
It seems this is actually an issue. The difference between my python implementation, which doesn't use cumsum at all, and GGIR, this is the error plot for ENMO over time. The one for EN is similar.

[image]

vincentvanhees, 2020-08-28:
Thanks, but I do not see the problem. The procedure is applied per 24 hours of data, your example shows that error does not get beyond e-12.

As of the new GGIR release (2.1-0) all aggregated values (ENMO, anglez, etc) are rounded to 4 decimal places before storages to lower size of output files and because anything beyond 4 decimal places is not of interest.