wadpac/GGIR #327: Avoidable NaN-values for angle{x,y,z}

prinsherbert, 2020-07-28:
Consider changing [these three expressions](https://github.com/wadpac/GGIR/blob/master/R/g.metric.R#L81) to:

    anglex = atan2(Gxm, sqrt(Gym^2 + Gzm^2)) / (pi/180)

to deal with the edge case that `Gym == Gzm == 0`, yet `Gxm != 0`, in which case the result should be `sign(Gxm) * 90`

vincentvanhees, 2020-07-28:
Thanks, will I incorporate this and your suggestion in issue #326 in upcoming release, which currently lives in this branch: https://github.com/wadpac/GGIR/tree/release2.1-0

Note that I have recently moved essential parts of g.getmetric to [g.applymetrics](https://github.com/wadpac/GGIR/blob/release2.1-0/R/g.applymetrics.R) and listed g.getmetric for depicration. In this way all raw data metrics are calculated in one place and in a way that no calculation are unnecessarily done twice.

I will try to clarify this better in the function documentation.

vincentvanhees, 2020-08-07:
I am now looking at this. R can handle tan(1/0) but gives an error message when I do atan(1,0).
So, this may be a Python specific feature of the atan function.

closed 2020-08-07 by vincentvanhees