wadpac/GGIR #333: Unwanted nights being included in part4 analysis

m-patterson, 2020-08-11:
On a local test bench of files, I am getting two nights included in part4 analysis that should not be using the latest v2.1-0 #330 

This is the same error as stated at the end of issue #321 discussion. On Vincent's machine, the night was not included, but on mine it is for some reason. The error is still happening on my machine.

Example 1:
[image]

Part4 nightsummary:
[image]

Part4 summary (all 6 nights in nightsummary are used):
[image]

Example 2:
[image]

Part4 nightsummary:
[image]

I'm worried that nights that are clearly not sleep are being included in the final analysis and are going to pollute the final, averaged values. 

Maybe, this is a small bug fix as it is, otherwise, perhaps, a check could be included that would look for at least some level of movement during the SPT for the night to be included. I would think it is reasonable to expect some movement if a person is wearing the sensor over a full night.

I was testing on the latest v2.1-0 release, but I've pushed the branch I was testing on to my local repo, just in case (release2.1-0_MP).

vincentvanhees, 2020-08-12:
Hi Matt, I see in your config.csv file that you have includenightcrit = 4, by which part 4 includes any night (noon-noon) with 4 or more valid hours. When I change that to includenightcrit= 16, which is normally the default, and re-run part 4, the first night is excluded.

However, this does not address the issue with night number 8, because that night has only 13% non-wear according to current non-wear detection (see column fraction_night_invalid). The threshold is 1- (16/24) = 33.33%. For night 8 it seems GGIR is underestimating the amount of non-wear, which I also see sometimes in other datasets. Possible solutions:
- If you use read.myacc.csv then maybe use a less stringent setting for rmc.dynamic_range, maybe the noise in your data is too close to the threshold that is used.
- Use 1800 instead of 3600 as third value in argument windowsizes, to make it use a 30 minute rolling window instead of 60 minute. This should increase sensitivity to non-wear.

The latter point may even need to be changed in the default argument values, but before we can do that I think it would be best if this is critically evaluated first. I will create a new issue for that now.

In #321 I focused on the double nights, which I think we resolved.

m-patterson, 2020-08-12:
Thanks for the feedback @vincentvanhees, that is very helpful. Yes, it was a mistake that I had left my _includenightcrit_ at 4 hours.

As per Example 2 mentioned above, great idea on creating a data-set to allow us to test performance on non-wear portions, I would be happy to work on that. There was a researcher at the conference in Spain who was working on this, I wonder if we should talk to him.

Also, I may have a quick look into implementing a quick 'artefact detection' check at the end of sleeponset/wakeup detection in order to prevent nights with very little movement through (in case the non-wear threshold doesn't cancel them). From your experience, do you think it is OK to set a minimum range of accel values on a nights sleep?

vincentvanhees, 2020-09-07:
### non-wear
Great that you have time to look into the non-wear detection.

I think [this](https://www.nature.com/articles/s41598-020-62821-2) work by Shaheen Syed from earlier this year could be informative (note that Shaheen is in OSS webinar series and presents later this year).

He  used ECG as ground truth and used the GGIR algortihm modified to be 30 minutes instead of 60 minutes, as that is how I initially presented it in 2011. Their positive results may be an indication that it is safe to switch from 60 to 30 minutes.

### artefact detection
Keep in mind that the day or night may still be of interest for research if a person does not wear the accelerometer during the sleep. For example, if a sleep diary is used and the sleep diary matches well with the non-wear window then a researcher may still be interested in using that night. Especially if their interest is primarily on time-use during the 24 hour day with part 5.

So, if you implement this then I would make it conditional on not having access to sleep diary.

Further, it may be easier to focus on improving the non-wear detection handle these cases then to on developing a new algorithm for detecting non-wear during the night.

vincentvanhees, 2020-09-25:
Looks like this issue has been addressed, so I am closing it now. Feel free to re-open if you feel this needs renewed attention.

closed 2020-09-25 by vincentvanhees