wadpac/GGIR #334: Default non-wear detection configuration

vincentvanhees, 2020-08-12:
I designed the heuristic non-wear detection algorithm that is part of GGIR to detect when a person does not wear the accelerometer for at least a couple of consecutive hours. The idea is that these are the most problematic types of non-wear from a research perspective. So, we take possible short episodes of non-wear for granted. For example, when people take the accelerometer off for a short period of time to prevent it from getting wet or dirty.

Sometimes we see that non-wear is missed by the algorithm when the accelerometer is not worn during the night, e.g. #333, which is not what we expect. 

**Possible explanation:**
- The threshold for detecting low standard deviation in raw signal is too low. This makes that non-wear is missed that involves a bit more than usual noise, vibrations (sensor not worn, but lying in bed), or temperature variations.
- The large window used by the algorithm, more specifically the 3rd value of input argument windowsizes, is too large. Maybe we  should change this value back to 30 minutes (1800 seconds) as in the 2011 publication as opposed to the 3600 seconds used since 2013. 

**To do:**
- Identify test files where the issue occurs.
- Try out shorter window size and see whether it helps.
- Experiment with shorter window size in large datasets and check whether this causes a big shift in detected non-wear time. If it does then a study will be needed to motivate why the shorter window size is better. Also it would be good to communicate with GGIR google group about this as GGIR users may be able to help investigate the impact of changing the argument value.

m-patterson, 2020-08-12:
This sounds like a great idea. If we want to go ahead with it, I think the first step would be set up a data-set with manual annotations indicating where non-wear periods occurred. This way we could try different thresholds and see the performance metrics on a wide range of tests, instead of just tuning the thresholds to specific tests. From my point of view, it would be OK for a researcher to annotate non-wear periods after the data has been recorded. I wonder if that researcher at the conference in Spain has data that would be possible for this?

Another factor to consider is the type of non-wear period. The most simple type would be when the sensor is taken off and does not move. In other use cases, non-wear might happen while the patient places the sensor in a bag, or leaves the sensor in their car (ie - instances where some movement can occur). Can we differentiate movement from wearing the sensor at the wrist (or hip) between other common non-wear scenarios? This might be a tough problem to solve considering the wide array of types of movements that patients perform in their daily lives during multi day recordings, especially sleep. In an ideal scenario, we would have a large enough test data-set with annotations on the 'type' of non-wear as well as when the non-wear periods occurred. This way, we could try to optimize thresholds to detect well on difficult to find non-wear periods without sacrificing performance on actual wear data.

m-patterson, 2020-09-04:
I'm currently developing a data-set of files that we can use as training and testing to optimize the non-wear algorithm parameters. One issue I am running into is that there is a small bit of motion during non-wear that causes the algorithm to miss the data around it. I'm not too sure if the best way forward is to change the annotations to account for these bits of sensor movement, or else to leave it as is. If I annotate around these movements, it feels like I am tuning the data-set to work better on the current algorithm. If I leave these movements, it will likely be performance around these locations which will dictate how well certain threshold combinations work. @vincentvanhees @jhmigueles what is your opinion? Do you think I should annotate the full section as non-wear, or make the short movement periods into wear time?
[image]

Here is a visualization of the performance with GGIR:
[image]

vincentvanhees, 2020-09-07:
I think it is good to label the entire period as non-wear, because this is also what we may see in real life data. A person may not wear the accelerometer, but still touches it from time to time.

The GGIR non-wear detection algorithm labels short lasting movements as nonwear if they are surrounded by large non-wear blocks. In the [supporting information](https://journals.plos.org/plosone/article?id=10.1371/journal.pone.0061691#s5) of the 2013 paper I documented this in more detail.  I suspect that the non-wear in the above example are not long enough to trigger this part of the algorithm. However, that also makes it a good test case to understand how we can minimize the incorrect classification in such cases. For example, would a 30 minute instead of 60 minute rolling window help.

m-patterson, 2020-09-14:
I've created a training data-set that we can test the 30 minute window on. There are 30 x 24hr worth of wrist data in this set and all the non-wear has been manually annotated after the recording was complete. Most 24hr periods contain a full night worth of sleep, as this seems to be an important area where we do not want false positives creeping in.

The GGIR non-wear algorithm was set up to run with different combinations of thresholds. A grid expand approach was used to test each combination, these were the thresholds tested in this initial run:
- Long period window: 1800, 3600
- Short period window: 600, 900
- SD criteria: 0.010, 0.013, 0.025
- Range criteria:  0.10, 0.15, 0.20

Since wear and non-wear periods were not balanced classes in this data, F1 score is the best metric to consider:
[image]

Long period window of 1800 and short period window of 600 resulted in the highest F1 score:
[image]

SD thres of 0.2 and Range Thres of 0.01 had the highest F1 score:
[image]

At the highest F1 score, false positives were at a local max and false negatives were at a local min:
[image]

Overall comments on the effect that different thresholds had on results:
- Decreasing the long window size always increased the F1 score. This is because the edges of non-wear periods are more classified as non-wear, however, there were also more false positives with the shorter long window size.
- Upon preliminary inspection, the Range Threshold had a large effect on F1 score, with each increase in Range Thres resulting in a higher F1 score
- Increasing SD Thres seemed to slightly decrease the F1 score upon preliminary inspection

Two questions for @vincentvanhees  that I have from this analysis are:
1.	Are there other threshold levels that you want to test out?
2.	Should we try to reduce false positives or false negatives? Or, just use the F1 score as is? I know F1 is suppose to balance FP and FN, but in this case the false negatives are low and the false positives are high at the peak F1 score. If we decide false positives are worse, then we could pick another threshold combination that had less false positives (higher precision)

vincentvanhees, 2020-09-15:
Thanks Matt, this is very helpful.

**Regarding results:**

> Decreasing the long window size always increased the F1 score. This is because the edges of non-wear periods are more classified as non-wear, however, there were also more false positives with the shorter long window size.

Sounds logical. On the other hand there is an increased risk that short episodes of rest are misclassified as non-wear, so I think it is important to let false positives weight heavily in the final score.

> Upon preliminary inspection, the Range Threshold had a large effect on F1 score, with each increase in Range Thres resulting in a higher F1 score

If I understand it correctly then this means that occasional spikes in the signal in combination with a low standard deviation over most likely to reflect non-wear. Does this then also mean that we can leave out the range criteria and fully rely on the standard deviation check on its own? In other words: Does F1 decrease if you set the range criteria very high, e.g. 2.0 _g_?

> Increasing SD Thres seemed to slightly decrease the F1 score upon preliminary inspection.

SD threshold optimization probably relates to dynamic range and resolution of the sensor, and maybe even sensor brand. So, it is probably better not to change this radically without testing how it affects data from other sensor brands.

**Regarding questions:**

> Are there other threshold levels that you want to test out?

Following my comment above, could you check whether a very high range threshold (2.0) negatively affects the performance? If not, then we could decide to remove this range check.

> Should we try to reduce false positives or false negatives? Or, just use the F1 score as is? I know F1 is suppose to balance FP and FN, but in this case the false negatives are low and the false [...]

False positive (mislabeling rest/sleep as non-wear) would be more concerning, because false negatives (mislabeling non-wear as wear) are likely to only occur if the non-wear is shorter than the detection window and by that unlikely to take up a large part of the recording. So, I think it is best to make the performance metric most sensitive to false positives.

**Other reflections:**

I did a few recordings where I left an Axivity sensor in my bed underneath my pillow for over a week. It would be interesting to see how this is treated by non-wear detection. My breathing and turning causes vibrations in the signals that may or may not trigger the wear detection.... I haven't checked this yet. It is understandable if performance is not always optimal, but on average I would hope that more than 80% of those nights are correctly classified as non-wear. Should we use this as additional test data? It may not be a common scenario, but it is not entirely unrealistic given that people may take off the sensor during sleep.  I can send you a copy of the file if you are interested.

m-patterson, 2020-09-15:
Thanks for the feedback @vincentvanhees, here are results with a larger range threshold and some other threshold adjustments:

Long window size: 1800, 3600, 4500 [4500 from Syed paper suggestion]
Short window size: 600, 900, 1200
SD-thres: 0.013
Range-thres: 0.1, 0.15, 0.2, 0.4, 2.0

Recall, precision and F1:
[image]

ws1 and ws2 thresholds at each testing iteration:
[image]

sd and range threshold at each testing iteration:
[image]

False negatives and false positives at each testing iteration:
[image]

Comments

- Increasing the long window size improves precision, but decreases recall and F1 (we care about precision the most in this application)
- Increasing the range threshold hurts precision (more false positives), but improves F1 via a large increase in recall score (less false negatives)

There are many combination of thresholds that have precision > 0.99:
[image]

Out of that list, the combinations with the top 3 F1 score are:
[image]

Short window size doesn't seem to have too much affect on the results. @vincentvanhees, do you think it would be worth while trying to make short window size the same as the large window size?

m-patterson, 2020-09-15:
Just for interests sake, I performed testing with the ideal thresholds, but made ws2 == ws1, here are the results:
[image]

False positive are still very low and precision is high, slightly more false negatives, as there would be a bigger chunk of any false detection that is labelled as wear with the larger small window size. If the testing is occurring on the large window size, do you think there is an argument to label the large window size instead of the small window size?

jhmigueles, 2020-09-15:
Thanks for the work, really interesting and helpful.

I think it is always difficult to find the perfect thresholds (either time windows or SD th) because they would change in different contexts (sensor brands, participants habits, etc). I agree the most relevant task in this classification problem is to discern between rest/sleep and non-wear. The SD and range thresholds are the most relevant for this, and it is difficult to test how they will behave in every sensor brand. Would it be possible to use the non-movement windows identified for auto-calibration to set device-specific SD and range thresholds?

vincentvanhees, 2020-09-21:
Thanks Matt. May I summarize from your overview that most performance improvement can be gained by increasing the range threshold from 0.15 to 0.2, and that sticking with ws1 = 3600 is advisable?

> Matt: "If the testing is occurring on the large window size, do you think there is an argument to label the large window size instead of the small window size"

I agree, but it would be good to keep the rolling window effect as that improves the resolution of the detection. For example, if an hour of non-wear starts at 5 minutes past the clock hour then we do not want to miss it because we only look at full clock hours. So, I envisage an update to the code where we keep a rolling ws1 window with steps of ws2, but where ws1 needs to align with the ws2 windows it covers. In other words: A 60 minute window that rolls over 10 minute windows is fine, but a 60 and 15 minute is not possible, because this would mean that the labelling becomes ambiguous.

> Jairo: "Would it be possible to use the non-movement windows identified for auto-calibration to set device-specific SD and range thresholds?"

In theory I agree, but it potentially results varying thresholds even within the same sensor brand. This may hard to interpret and could cause concern about the reliability of the non-wear detection. So, I think this would have to be carefully evaluated and compared with controlled experiments of the noise in various sensor brands to reassure user that this always works flawlessly. I think you need large cohort data for this to make a convincing case that even in thousands of measurements errors never occur.

m-patterson, 2020-09-21:
The optimal thresholds that I discovered from the work were, large window = 3600, medium window = 1200, range-thres = 0.2 and SD-thres = 0.013. It should be noted that this was only done with Verisense data, so may not be applicable to other hardware.

> Thanks Matt. May I summarize from your overview that most performance improvement can be gained by increasing the range threshold from 0.15 to 0.2, and that sticking with ws1 = 3600 is advisable?

Yes, the large window size of 3600 was the most appropriate. Precision was already very high with the default range threshold of 0.15, but with the increase to 0.2, the precision remained high and the recall (false negative rate) increased as well, increasing the F1-score. Since this analysis was done with Verisense data only, I'll look at implementing an option to use 0.2 range-thres for Verisense data.

> I agree, but it would be good to keep the rolling window effect as that improves the resolution of the detection. 

Agreed, the smaller window size to move forward is important, as 1-hour non-overlapping windows would definitely miss non-wear. However, by only labelling the medium window size as non-wear, we know that we are going to mis-classify the start and the end of each non-wear segment as wear, like this:
[image]

> A 60 minute window that rolls over 10 minute windows is fine, but a 60 and 15 minute is not possible, because this would mean that the labelling becomes ambiguous.

Isn't 15 minute the default window size in GGIR (900 seconds)? I don't think I understand what the issue is with overlap in the labelling?  Would it not be possible to label the full 3600 sec (full hour) as non-wear, but still move the large window forward by 10 or 15min increments? Yes, some sections will be labelled over several times, but could the logic state that if at least one iteration of non-wear detection states non-wear, then that time period is non-wear (further iterations at the time point can say wear or non-wear, it won't matter once a time period is labelled once). This is just a thought, as from my point of view the medium window size is important so we don't move the large window along too much, but I do not see the point of only labelling a small portion in the middle of the large window as non-wear after all the data in the large window passed the checks for non-wear. This way, non-wear labelling would bump right up against wear time.

vincentvanhees, 2020-09-22:
> I don't think I understand what the issue is with overlap in the labelling? 

Currently the large window is centered at the center of the short window. For a 15 and 60 minute window combination, this would mean that there are 22.5 minutes on the left of the 15 minute window an 22.5 minutes on the right of the window, which makes it unclear how to label the underlying 15 minute windows.

If we do not want to restructure the existing code and milestone data structure too much then I see as possible solutions:
1. Change how the large and short window are aligned. For example, the large window simply starts when short window starts. So, if large window detects non-wear than the underlying short windows will be tagged as non-wear.
2. Require that large window is an odd multitude of the short window length, because only then we can keep the large window centered in the short window.

In terms of non-wear detection quality it should make no difference, it is mainly a matter of how to modify the code efficiently.

vincentvanhees, 2020-12-04:
I am now closing this issue, feel free to re-open if you think this should remain on the agenda.

closed 2020-12-04 by vincentvanhees