wadpac/GGIR #358: GGIR logo requirements

vincentvanhees, 2020-09-22:
I am talking to a graphics designer to design a logo for GGIR. Here are the requirements I have in mind. 

Feedback welcome! @m-patterson @jhmigueles ?

**Intended application of the logo:**
- Online documentation.
- Laptop sticker
- Used inside PowerPoint presentations.
- On my own website and possibly other websites

**Text:**
GGIR (in capitals)
Subtitle: R package for accelerometer data analysis.

**Lay man summary of what GGIR is, such that designer can keep this in mind:**
GGIR is free software  to analyse data collected with wearable movement sensors.
For example, it can detect how long a person has slept or how active a person has been.
GGIR is primarily used by movement- and sleep researchers, passionate about understanding and improving human health.

**Shape / Colour preferences:**
I do not have a strong preference. The style should be distinct from my Accelting logo to avoid the impression that GGIR is an exclusive product by me. In terms of shape: We should not copy the tidyverse hectogon, because GGIR is not tidyverse. However any other simple but recognisable shape could work.

**What are core strengths of GGIR that we want the logo to radiate?**
- Broad spectrum of functionalities / Modifiability
- Open / Transparent
- Control for researcher
- Ease of use (despite the learning curve, you can learn GGIR without prior programming experience)
- Evidence based
- ...

vincentvanhees, 2020-09-22:
It could be that the designer complains that the subtitle 'R package for accelerometer data analysis.' is too long.
In that case we could maybe shorten it to 'R package for accelerometry'.

Not using a subtitle has the risk that the logo becomes incomprehensible when presented without context. This is not a problem all the time, e.g. when it is used in our package documentation... but it is generally helpful to have a logo that is self-explanatory.

m-patterson, 2020-09-22:
I think an important strength of GGIR is that it is used to get useful insights from remotely captured health data, which is pretty important in the context of the world today, so I would suggest adding the following bullet point to the 'core strengths' section:

- Analyze remotely captured health data

jeremy-dai, 2020-09-24:
[image]

vincentvanhees, 2020-09-25:
Thanks @jeremy-dai :) I am actually hiring a designer to work on this, but it is good to know that you are looking at alternatives.

vincentvanhees, 2021-01-06:
Closing this issue, because we have a logo now. I am now going to open a separate issue for integrating this in GGIR.

closed 2021-01-06 by vincentvanhees