wadpac/GGIR #368: Axivity AX3 - Error in the first data block reading

GeweiZhu, 2020-10-20:
Hi,

I'm quite new to GGIR and I have encountered a problem recently. I processed a couple of cwa files from Axivity AX3 accelerometer, but one of them wouldn't process. An error message showed up stating "Error in readHeader (fid, numDBlocks): Error in the first data block reading". But I have used the same codes for all files and it worked fine for other files. Also I managed to open it in OmGui and it doesn't seem to have a problem. I can't upload the cwa file here as it's too big. Please let me know what would be the best way to share the file with you.  I would appreciate any advice on this matter.
Thank you for your help in advance.

Becky

Mirkes, 2020-10-20:
Dear Becky,

Could you send me the first 100 kilobytes of your cwa file? It looks like the problem in the beginning of file and whole file is not necessary.

Best regards,
Evgeny

GeweiZhu, 2020-10-20:
> ⚠ External sender. Take care when opening links or attachments. Do not provide your login details.

Hi Evgeny,

Thank you for your reply.
Sorry, I'm really new to accelerometer, too. Therefore, I'm not sure how to only send part of my cwa file.

Best wishes
Becky

Gewei Zhu (Becky)

PhD student

Dermatological Sciences

Institute of Translational and Clinical research

2nd floor, William Leech building

Medical school, Framlington place

Newcastle University

NE2 4HH

Mirkes, 2020-10-20:
Dear Becky,

Sorry but I do not know your OS.

This link is useful for Linux: https://www.networkworld.com/article/3489256/breaking-linux-files-into-pieces-with-the-split-command.html#:~:text=To%20split%20a%20file%20into,simply%20use%20the%20split%20command.&text=By%20default%2C%20the%20split%20command,chunks%20named%20xza%20and%20xzz.

This link is for Windows: https://superuser.com/questions/94083/how-to-split-large-file-on-windows

This is for MacOS https://superuser.com/questions/313014/how-do-i-use-split-in-the-mac-os-x-terminal-to-split-files-directly-to-another-d

Best regards,
Evgeny

vincentvanhees, 2020-10-20:
dear both, this issue may overlap with another issue I very recently fixed.

You can find a description of that issue here: https://github.com/wadpac/GGIR/issues/361
The pull request that will fix it is here: https://github.com/wadpac/GGIR/pull/367
I plan to merge this in the master branch later today.

To use this version of GGIR do:

```
install.packages("devtools")
library(devtools)
devtools::install_github("wadpac/GGIR")

```
restart R(Studio)

`library(GGIR)`

Mirkes, 2020-10-20:
Thanks Vincent, It is possible.

Dear Becky,

Could you check proposed way?

Best regards,
Evgeny

GeweiZhu, 2020-10-20:
Hi Evgeny and Vincent,

Thank you both for your advice. I will give it a go later as my laptop is currently frozen after I tried to split my cwa file.

Best wishes
Becky

Gewei Zhu (Becky)

PhD student

Dermatological Sciences

Institute of Translational and Clinical research

2nd floor, William Leech building

Medical school, Framlington place

Newcastle University

NE2 4HH

vincentvanhees, 2020-12-04:
I am now closing this issue now, but feel free to re-open it if the problem persists.

closed 2020-12-04 by vincentvanhees