wadpac/GGIR #369: CWA difference

muschellij2, 2020-10-21:
I'm going to open one here and at @activityMonitoring as well for https://github.com/activityMonitoring/biobankAccelerometerAnalysis.

Below I show 2 things: 
1) some really weird values from GGIR (way too high)
2) Skips in python3 reading from @activityMonitoring

    # python3 accProcess.py --skipCalibration=True --rawOutput=True data/example_90001_0_0.cwa.gz

    library(readr)
    library(GGIR)
    library(dplyr)
    #> 
    #> Attaching package: 'dplyr'
    #> The following objects are masked from 'package:stats':
    #> 
    #>     filter, lag
    #> The following objects are masked from 'package:base':
    #> 
    #>     intersect, setdiff, setequal, union
    fname = "data/example_90001_0_0.cwa.gz"
    tfile = R.utils::gunzip(fname, 
                            remove = FALSE, temporary = TRUE)

    out = GGIR::g.cwaread(tfile, end = Inf)
    data = out$data[, c("time", "x", "y", "z")]
    data$time = as.POSIXct(data$time, origin = "1970-01-01")
    dim(data)
    #> [1] 29864812        4
    data[22656010:22656030,]
    #>                         time        x         y          z
    #> 22656010 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 22656011 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 22656012 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 22656013 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 22656014 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 22656015 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 22656016 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 22656017 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 22656018 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 22656019 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 22656020 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 22656021 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 22656022 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 22656023 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 22656024 2015-06-22 00:56:04 -0.50000  0.359375   0.609375
    #> 22656025 2015-06-22 00:56:04 48.11477 76.729466 123.842931
    #> 22656026 2015-06-22 00:56:04 48.09753 76.702373 123.799213
    #> 22656027 2015-06-22 00:56:04 48.08029 76.675281 123.755496
    #> 22656028 2015-06-22 00:56:04 48.06305 76.648189 123.711779
    #> 22656029 2015-06-22 00:56:04 48.04581 76.621096 123.668062
    #> 22656030 2015-06-22 00:56:04 48.02857 76.594004 123.624344

    csv = readr::read_csv(sub("cwa", "csv", fname))
    #> 
    #> ── Column specification ────────────────────────────────────────────────────────
    #> cols(
    #>   time = col_character(),
    #>   x = col_double(),
    #>   y = col_double(),
    #>   z = col_double()
    #> )
    dim(csv)
    #> [1] 29856000        4
    head(csv)
    #> # A tibble: 6 x 4
    #>   time                                              x     y      z
    #>   <chr>                                         <dbl> <dbl>  <dbl>
    #> 1 2015-06-19 10:00:03.771+0100 [Europe/London] -0.281 0.781 -0.75 
    #> 2 2015-06-19 10:00:03.781+0100 [Europe/London] -0.422 0.25   0.703
    #> 3 2015-06-19 10:00:03.791+0100 [Europe/London] -0.422 0.281  0.734
[... 72 lines cut ...]
    #>  R.oo          1.24.0     2020-08-26 [1] CRAN (R 4.0.2)                   
    #>  R.utils       2.10.1     2020-08-26 [1] CRAN (R 4.0.2)                   
    #>  R6            2.4.1      2019-11-12 [2] CRAN (R 4.0.0)                   
    #>  Rcpp          1.0.5      2020-07-06 [2] CRAN (R 4.0.0)                   
    #>  readr       * 1.4.0      2020-10-05 [1] CRAN (R 4.0.2)                   
    #>  reprex        0.3.0.9001 2020-09-30 [1] Github (tidyverse/reprex@d3fc4b8)
    #>  rlang         0.4.8.9000 2020-10-20 [1] Github (r-lib/rlang@011cb4c)     
    #>  rmarkdown     2.4        2020-09-30 [1] CRAN (R 4.0.2)                   
    #>  rstudioapi    0.11       2020-02-07 [2] CRAN (R 4.0.0)                   
    #>  sessioninfo   1.1.1      2018-11-05 [2] CRAN (R 4.0.0)                   
    #>  stringi       1.5.3      2020-09-09 [1] CRAN (R 4.0.2)                   
    #>  stringr       1.4.0      2019-02-10 [2] CRAN (R 4.0.0)                   
    #>  styler        1.3.2      2020-02-23 [2] CRAN (R 4.0.0)                   
    #>  tibble        3.0.4      2020-10-12 [1] CRAN (R 4.0.2)                   
    #>  tidyselect    1.1.0      2020-05-11 [2] CRAN (R 4.0.0)                   
    #>  utf8          1.1.4      2018-05-24 [2] CRAN (R 4.0.0)                   
    #>  vctrs         0.3.4      2020-08-29 [1] CRAN (R 4.0.2)                   
    #>  withr         2.3.0      2020-09-22 [1] CRAN (R 4.0.2)                   
    #>  xfun          0.18       2020-09-29 [1] CRAN (R 4.0.2)                   
    #>  yaml          2.2.1      2020-02-01 [2] CRAN (R 4.0.0)                   
    #> 
    #> [1] /Users/johnmuschelli/Library/R/4.0/library
    #> [2] /Library/Frameworks/R.framework/Versions/4.0/Resources/library

</details>

muschellij2, 2020-10-21:
I added the output from `cwa-convert` from https://github.com/digitalinteraction/openmovement/tree/master/Software/AX3/cwa-convert to show that even the first 3 rows  have differences:

    # python3 accProcess.py --skipCalibration=True --rawOutput=True data/example_90001_0_0.cwa.gz

    setwd("~/Dropbox/Packages/biobankAccelerometerAnalysis/")
    library(readr)
    library(GGIR)
    library(dplyr)
    #> 
    #> Attaching package: 'dplyr'
    #> The following objects are masked from 'package:stats':
    #> 
    #>     filter, lag
    #> The following objects are masked from 'package:base':
    #> 
    #>     intersect, setdiff, setequal, union
    fname = "data/example_90001_0_0.cwa.gz"
    xyz = c("x", "y", "z")
    tfile = R.utils::gunzip(fname, 
                            remove = FALSE, temporary = TRUE)

    out = GGIR::g.cwaread(tfile, end = Inf)
    data = out$data[, c("time", xyz)]
    data = tibble::as_tibble(data)
    data$time = as.POSIXct(data$time, origin = "1970-01-01")
    dim(data)
    #> [1] 29864812        4
    as.data.frame(data[22656010:22656030,])
    #>                   time        x         y          z
    #> 1  2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 2  2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 3  2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 4  2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 5  2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 6  2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 7  2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 8  2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 9  2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 10 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 11 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 12 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 13 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 14 2015-06-22 00:56:03 -0.50000  0.359375   0.609375
    #> 15 2015-06-22 00:56:04 -0.50000  0.359375   0.609375
    #> 16 2015-06-22 00:56:04 48.11477 76.729466 123.842931
    #> 17 2015-06-22 00:56:04 48.09753 76.702373 123.799213
    #> 18 2015-06-22 00:56:04 48.08029 76.675281 123.755496
    #> 19 2015-06-22 00:56:04 48.06305 76.648189 123.711779
    #> 20 2015-06-22 00:56:04 48.04581 76.621096 123.668062
    #> 21 2015-06-22 00:56:04 48.02857 76.594004 123.624344

    csv = readr::read_csv(sub("cwa", "csv", fname))
    #> 
    #> ── Column specification ────────────────────────────────────────────────────────
    #> cols(
    #>   time = col_character(),
    #>   x = col_double(),
    #>   y = col_double(),
    #>   z = col_double()
    #> )
    dim(csv)
    #> [1] 29856000        4
    head(csv)
    #> # A tibble: 6 x 4
    #>   time                                              x     y      z
    #>   <chr>                                         <dbl> <dbl>  <dbl>
    #> 1 2015-06-19 10:00:03.771+0100 [Europe/London] -0.281 0.781 -0.75 
    #> 2 2015-06-19 10:00:03.781+0100 [Europe/London] -0.422 0.25   0.703
    #> 3 2015-06-19 10:00:03.791+0100 [Europe/London] -0.422 0.281  0.734
    #> 4 2015-06-19 10:00:03.801+0100 [Europe/London] -0.406 0.297  0.734
[... 180 lines cut ...]
    #>  R.oo          1.24.0     2020-08-26 [1] CRAN (R 4.0.2)                   
    #>  R.utils       2.10.1     2020-08-26 [1] CRAN (R 4.0.2)                   
    #>  R6            2.4.1      2019-11-12 [2] CRAN (R 4.0.0)                   
    #>  Rcpp          1.0.5      2020-07-06 [2] CRAN (R 4.0.0)                   
    #>  readr       * 1.4.0      2020-10-05 [1] CRAN (R 4.0.2)                   
    #>  reprex        0.3.0.9001 2020-09-30 [1] Github (tidyverse/reprex@d3fc4b8)
    #>  rlang         0.4.8.9000 2020-10-20 [1] Github (r-lib/rlang@011cb4c)     
    #>  rmarkdown     2.4        2020-09-30 [1] CRAN (R 4.0.2)                   
    #>  rstudioapi    0.11       2020-02-07 [2] CRAN (R 4.0.0)                   
    #>  sessioninfo   1.1.1      2018-11-05 [2] CRAN (R 4.0.0)                   
    #>  stringi       1.5.3      2020-09-09 [1] CRAN (R 4.0.2)                   
    #>  stringr       1.4.0      2019-02-10 [2] CRAN (R 4.0.0)                   
    #>  styler        1.3.2      2020-02-23 [2] CRAN (R 4.0.0)                   
    #>  tibble        3.0.4      2020-10-12 [1] CRAN (R 4.0.2)                   
    #>  tidyselect    1.1.0      2020-05-11 [2] CRAN (R 4.0.0)                   
    #>  utf8          1.1.4      2018-05-24 [2] CRAN (R 4.0.0)                   
    #>  vctrs         0.3.4      2020-08-29 [1] CRAN (R 4.0.2)                   
    #>  withr         2.3.0      2020-09-22 [1] CRAN (R 4.0.2)                   
    #>  xfun          0.18       2020-09-29 [1] CRAN (R 4.0.2)                   
    #>  yaml          2.2.1      2020-02-01 [2] CRAN (R 4.0.0)                   
    #> 
    #> [1] /Users/johnmuschelli/Library/R/4.0/library
    #> [2] /Library/Frameworks/R.framework/Versions/4.0/Resources/library

</details>

Mirkes, 2020-10-21:
It is necessary to understand that cwa file contains records with unfixed frequency. I do not know why but it is fact. As a result all software must use some procedures to provide normal discretisation frequency. In cwareader the linear interpolation is used to convert data to the 100Hz.
Another software can use different method.

Hope it is useful.

Best regards,
Evgeny

vincentvanhees, 2020-10-21:
Thanks @muschellij2 - this issue may overlap with #361 which I closed last week after making pull request #367. The issue was that sample frequency was not correctly recognised from the page header..

I see you are using an older GGIR version, does the issue persist when you use the GGIR master branch here on GitHub?

In January I expanded the g.cwaread functionality to also be able to handle AX6 cwa file formats and with help from Dan Jackson we resolved some issues in the numUnpack function affecting AX3 data with non-default configuration settings.

vincentvanhees, 2020-10-21:
just to re-assure any PA/Sleep researcher reading this. The GGIR code does flag every large QC window (default 15 minutes)  as invalid if it has even a single raw acceleration value larger than dynamic range x 1.5, positive or negative. This as a safety net in case something goes wrong with reading the data or files get corrupted. That said, it would be good to investigate what is going on.

vincentvanhees, 2020-10-22:
>  ... even the first 3 rows have differences ...

Different packages may handle start of the file differently. For example, there is a known issue with AX3 values in some specific firmware versions being 'stuck' for the first couple of seconds. Some packages may intentionally skip these values while other package simply include them in the output.  So, I think it is best not to rely too heavily on row or page number but focus on matching timestamps instead. 

@muschellij2 if you could also send me a copy of the problematic file that would be much appreciated. In the mean time I am trying to retrieve a copy of a file from another group who had a similar problem a while ago, which I now realize could have been the same issue.

vincentvanhees, 2020-10-22:
1. Where can I find accProcess.py?

> The last CSV is from cwaconvert https://github.com/digitalinteraction/openmovement/tree/master/Software/AX3/cwa-convert

2. which of the cwaconvert tools in that directory did you use?

muschellij2, 2020-10-22:
>    1. which of the cwaconvert tools in that directory did you use?

accProcess.py is at
https://github.com/activityMonitoring/biobankAccelerometerAnalysis
In
https://github.com/digitalinteraction/openmovement/tree/master/Software/AX3/cwa-convert
I used c/, ran make, then used cwa-convert from that.
Best,
John

vincentvanhees, 2020-10-22:
### minor differences in acceleration values

When I plot the extracted acceleration signal derived with OMGUI csv export which internally uses cwaconvert and compare this with GGIR::g.cwaread output I see a small, approximately constant, offset between the two signals which otherwise follow the same pattern in time, which seems to match John's observation

This evening I reviewed all the code in GGIR's [g.cwaread](https://github.com/wadpac/GGIR/blob/master/R/g.cwaread.R) and [numUnpack](https://github.com/wadpac/GGIR/blob/master/src/numUnpack.cpp) relative to the cwa file format documentation in https://github.com/digitalinteraction/openmovement/blob/master/Docs/ax3/cwa.h
and I could not find any obvious discrepancies.

The part of the code I found more difficult to review is the resampling step because this is not cwa format specific:
[GGIR's resampling function](https://github.com/wadpac/GGIR/blob/master/src/resample.cpp) looks similar to what is done in [OMGUI](https://github.com/digitalinteraction/openmovement/blob/6233f335babb9918bbd3ad84ea5011692cb02ab6/Software/AX3/omconvert/omconvert.c)**(see LinearInterpolate function)** although I am not entirely sure this is the actual code used by cwaconvert and I am also not familiar with c, so maybe I am misjudging this. 

The resampling in [biobankAccelerometerAnalysis](https://github.com/activityMonitoring/biobankAccelerometerAnalysis/blob/6f6a60c2838706f52526dfe0af8c4551c168332d/java/Resample.java) looks different from what GGIR does....or is it mathematically equivalent?

@Mirkes would you mind having a look at these resample functions and check whether you can spot differences that could explain the minor differences in acceleration values?

### extreme values

In current version of GGIR these are flagged and ignored (not g.cwaread but more upstream in GGIR). Hopefully it will be possible to retrieve the problematic example file in the upcoming weeks to investigate further.

muschellij2, 2020-10-22:
I've requested the ability to share the UK Biobank file from the UKBB team.

vincentvanhees, 2020-10-23:
@danielgjackson Is there is a difference in cwa file format between (1) the cwa files used in UK Biobank and the [demofile](http://www.axivity.com/files/resources/longitudinal_data.zip) on the Axivity website and (2) cwa files extracted from commercially sold AX3 sensors? 

Commercial AX3 sensor output is identical across OMGUI exported csv files and GGIR. The small offset in values seems specific to UK Biobank data and the Axivity demofile.

Further, I am think the resampling inside GGIR is definitely not the problem, because when I turn it off and compare output with GGIR-based resampled data the signal aligns well as it should.

Mirkes, 2020-10-23:
Dear all,

When I developed resampling for cwaread function I used the very simple linear interpolation algoritm.
I know that open code software presents two different algorithms for java and for python implementation.
Unfortunately there is no unique fixed “true” algorithm for such resampling.

Best regards,
Evgeny

muschellij2, 2020-10-23:
I don't think the resampling is the problem.  I wrapped that software btw to https://github.com/muschellij2/cwaconvert so that you can run:
```r
remotes::install_github("muschellij2/cwaconvert")
library(cwaconvert)
out = read_cwa(file)
```

muschellij2, 2020-10-28:
I don't think the resampling is the problem.  I wrapped that software from @activityMonitoring btw to https://github.com/muschellij2/pycwa so that you can run:
```r
remotes::install_github("muschellij2/pycwa")
library(pycwa)
out = pyread_cwa(file)
```
@aidendoherty

vincentvanhees, 2020-10-28:
Thanks @muschellij2. For my own cwa test file collected with commercial AX3 sensor the `pycwa` function output looks consistent with `g.cwaread` and `omgui` export. So, this indicates that the observed small offset in raw data truly is specific to the combination of `GGIR::g.cwaread` and older cwa data. I am currently not working on paid projects related to cwa data but am happy to invest a bit more time on it once someone has been able to pinpoint the cause of the issue (or contracts me to dedicate more time to it). Hopefully my input here has at least been helpful to inform further investigation by others.

muschellij2, 2020-10-30:
So I made 2 other package for reading CWA files: https://github.com/muschellij2/pycwa and https://github.com/muschellij2/read.cwa for comparison to GGIR and potential checks.

    setwd("~/Dropbox/Packages/biobankAccelerometerAnalysis/")
    library(readr)
    library(GGIR)
    library(dplyr)
    #> 
    #> Attaching package: 'dplyr'
    #> The following objects are masked from 'package:stats':
    #> 
    #>     filter, lag
    #> The following objects are masked from 'package:base':
    #> 
    #>     intersect, setdiff, setequal, union
    library(read.cwa)
    library(pycwa)
    library(lubridate)
    #> 
    #> Attaching package: 'lubridate'
    #> The following objects are masked from 'package:base':
    #> 
    #>     date, intersect, setdiff, union
    file = "data/example_90001_0_0.cwa"

    xyz = c("x", "y", "z")
    py = py_read_cwa(file, verbose = FALSE)

    om = read_cwa(file, verbose = FALSE)
    gg = GGIR::g.cwaread(file, end = Inf, progressBar = TRUE, desiredtz = "UTC")
    #>   |                                                                              |                                                                      |   0%  |                                                                              |                                                                      |   1%  |                                                                              |=                                                                     |   1%  |                                                                              |=                                                                     |   2%  |                                                                              |==                                                                    |   2%  |                                                                              |==                                                                    |   3%  |                                                                              |==                                                                    |   4%  |                                                                              |===                                                                   |   4%  |                                                                              |===                                                                   |   5%  |                                                                              |====                                                                  |   5%  |                                                                              |====                                                                  |   6%  |                                                                              |=====                                                                 |   6%  |                                                                              |=====                                                                 |   7%  |                                                                              |=====                                                                 |   8%  |                                                                              |======                                                                |   8%  |                                                                              |======                                                                |   9%  |                                                                              |=======                                                               |   9%  |                                                                              |=======                                                               |  10%  |                                                                              |=======                                                               |  11%  |                                                                              |========                                                              |  11%  |                                                                              |========                                                              |  12%  |                                                                              |=========                                                             |  12%  |                                                                              |=========                                                             |  13%  |                                                                              |=========                                                             |  14%  |                                                                              |==========                                                            |  14%  |                                                                              |==========                                                            |  15%  |                                                                              |===========                                                           |  15%  |                                                                              |===========                                                           |  16%  |                                                                 [... cut ...]
    #>  utf8          1.1.4      2018-05-24 [2] CRAN (R 4.0.0)                   
    #>  vctrs         0.3.4      2020-08-29 [1] CRAN (R 4.0.2)                   
    #>  withr         2.3.0      2020-09-22 [1] CRAN (R 4.0.2)                   
    #>  xfun          0.18       2020-09-29 [1] CRAN (R 4.0.2)                   
    #>  yaml          2.2.1      2020-02-01 [2] CRAN (R 4.0.0)                   
    #> 
    #> [1] /Users/johnmuschelli/Library/R/4.0/library
    #> [2] /Library/Frameworks/R.framework/Versions/4.0/Resources/library

</details>

danielgjackson, 2020-11-02:
> @danielgjackson Is there is a difference in cwa file format between (1) the cwa files used in UK Biobank and the [demofile](http://www.axivity.com/files/resources/longitudinal_data.zip) on the Axivity website and (2) cwa files extracted from commercially sold AX3 sensors?

There shouldn't be any difference that would explain an offset.  

Very early firmware could not measure fractional timestamps, but did indicate the sample on which the second rolled over, this allowed recreation of timestamps of roughly the sample interval.  Fractional timestamps were added in a relatively complicated way so that it was backwards-compatible with this method.  Perhaps this code path for non-fractional times is somehow different in GGIR?  

> The small offset in values only seems to appear in UK Biobank data and the Axivity demofile

If the above does not turn up anything useful, I think this really must be a "raw export" (cwa-convert) comparison to really see what's going on.

Also: Biobank data is in the "packed" mode, so it's worth ensuring that any comparison is on the same basis.

vincentvanhees, 2020-12-23:
I have had another look at this issue and have come to believe that the issue is not in GGIR::g.cwaread but in the resampling algorithm as implemented in OMGUI and probably also the resampling step of biobankAccelerometerAnalysis.

**Motivation:**
To demonstrate this I used OMGUI to export the .cwa AX3 [demofile from the Axivity website](http://www.axivity.com/files/resources/longitudinal_data.zip) to three different .csv file types:
- File1 - With resampling in _g_-units
- File2 - Without resampling in _g_-units
- File3 - Without resampling in raw values (1-256).

When I plot the content of File1 and File2 on top of each other I get the exact same offset as I showed earlier in this thread. So, the offset does not originate from GGIR::g.cwaread but from OMGUI's own resampling step @danielgjackson. As biobankAccelerometerAnalysis had consistent output with OMGUI this may indicate that the same issue applies there too. @aidendoherty: To test this you will have to plot the resampled and original acceleration signal on top of each other as a function of timestamp.

GGIR's resampling code is [here](https://github.com/wadpac/GGIR/blob/master/src/resample.cpp).

**Unrelated extra checks on GGIR**
Following Dan's advice I also looked closer at GGIR. I edited GGIR::g.cwaread to not scale the acceleration signal such that it exports only the raw unscaled values. Next, I plotted this signal on top of the content of File3, which looks identical. So, the raw value extraction seems to be working fine in GGIR::g.cwaread, even for these older cwa-recordings. Further, I see that GGIR::g.cwaread correctly identifies the scaling factor to be 4096.

vincentvanhees, 2021-01-06:
I am now closing this issue:
- I have created a separate issue for the extreme values issue #380. I think that helps to focus follow-up efforts.
- Differences in the start of the recordings are expected in the order of seconds. What matters most is that patterns in data match when plotted against timestamps. There we see that data is in sync across the software tools (see plot earlier in the thread), which is good. 
- I think the offset in values between the software tools is explained now (see my previous post above). I will open a separate issue in openmovement to help make sure this receives attention.

closed 2021-01-06 by vincentvanhees