wadpac/GGIR #402: g.create.sp.mat (in g.part4)

RubenBrondeel, 2021-03-01:
the function g.create.sp.mat returns the calendar date to identify the night. It usually uses the first observation to determine the calendar date (calendar_date_safe). However, when the night goes from saturday to sunday, the weekday (wday) of the first observation is higher than the weekday of the last observation (7, and 1). In that case, the last date is returned as the calendar date. This results in nights with the same calendar dates, in which case the function g.part5 results in an error. (see below)

Errors and warnings for C1_305_PRE_44878.cwa.RData$message
[1] "number of items to replace is not a multiple of replacement length"

$call
dsummary[di, fi:(fi + 5)] <- c(summarysleep_tmp2$night[dayofinterst], 
    summarysleep_tmp2$daysleeper[dayofinterst], summarysleep_tmp2$cleaningcode[dayofinterst], 
    summarysleep_tmp2$guider[dayofinterst], summarysleep_tmp2$sleeplog_used[dayofinterst], 
    summarysleep_tmp2$acc_available[dayofinterst])

vincentvanhees, 2021-03-15:
While trying to reproduce the error I encountered some issues with function `g.sib.sum` (called from `g.part3`).
I have now committed some changes that should address this.

After this I was not able to replicate the error any more. Can you check that the error is also fixed for you when you use the branch `issue404_dayborder_impacton_sleep`? 

`library(remotes)`
`remotes::install_github("wadpac/GGIR", ref = "issue404_dayborder_impacton_sleep")`

On a side note - "C1_305_PRE_44878.cwa.RData" is generally a challenging file. The few available valid days are none-consecutive which complicates a reliable detection of sleep-wake rhythms in GGIR part5.

vincentvanhees, 2021-03-28:
I am now closing this issue, feel free to re-open if the problem comes back.

closed 2021-03-28 by vincentvanhees