wadpac/GGIR #407: using .csv file - error in g.part1

RubenBrondeel, 2021-03-24:
Using g.shell.GGIR or g.part 1 function on a .csv file, I got following error. When using read.myacc.csv(), there is no problem. 

Error in makePSOCKcluster(names = spec, ...) : 
  Cluster setup failed. 3 of 3 workers failed to connect.

When using a filename instead of de folder name, I got this error:
Error in if (config[ci, 2] == "c()") { : 
  missing value where TRUE/FALSE needed

I can't seem to find the origin of the error myself. One thing that did trow back errors when I was trying to debug g.part1, is the lack of origin in as.POSIXlt() (in g.inspectfile, g.getmeta and get_starttime_weekday_meantemp_truncdata)

vincentvanhees, 2021-04-26:
> Error in makePSOCKcluster(names = spec, ...) :
> Cluster setup failed. 3 of 3 workers failed to connect.

Not sure if it is related, but GGIR currently uses the maximum possible. We may need to add option for GGIR user to specify maximum number of process to use. I noticed that this may be relevant when working on large shared machines with many cores. 

> When using a filename instead of de folder name, I got this error:
> Error in if (config[ci, 2] == "c()") { :
> missing value where TRUE/FALSE needed

This looks like there is an issue with the config file. Can you try to delete the file and first process the data without specifying a config file?

>  the lack of origin in as.POSIXlt() 

I have not seen this issue before, I assumed that it would fall back on the default 1-1-1970. Could it be that this is not the default on every machine?

angerhang, 2021-05-11:
In fact, I am also experiencing similar issues on Mac machines with both Intel and M1 chips:
```R
library('GGIR')
g.shell.GGIR(datadir = "~/Desktop/tmp_data/my_data/",
             outputdir = "~/Desktop/tmp_data/my_results/",
             multicore = FALSE,
             mode=c(1:5), do.report=c(2,4,5))
```
The cwa file that I use is the sample cwa file that is being used from https://github.com/activityMonitoring/biobankAccelerometerAnalysis

Not sure what's going on (error msg from intel machine):
```
Error in makePSOCKcluster(names = spec, ...) : 
  Cluster setup failed. 7 of 7 workers failed to connect.
```

On M1 machine, Rstuio will just run for an hour without returning anything. Maybe a timeout should be used and raise a warning to the user given the size of the file.

vincentvanhees, 2021-05-11:
Just did a quick online search for your error message and it seems that this could also be a more fundamental issue with RStudio / parallel package on Mac. Either way, I am curious whether my suggestion above with argument `maxNcores` could be a workaround.

vincentvanhees, 2021-05-12:
Hi Hang, I suspect that this relates to something outside GGIR and possibly specific to Mac. I do not have a Mac so this is difficult for me to reproduce.

1. Would you mind checking `sessionInfo()` and confirm that:
- You have `foreach_1.5.1` or higher?
- You have `doParallel_1.0.16` or higher?
- You have R version `4.0.5` or higher?
- You have RStudio version `1.3.1093` or higher?

2. Can you run the following code to see whether this reproduces the error:

```
library("foreach")
library("doParallel")
cl <- parallel::makeCluster(1)
doParallel::registerDoParallel(cl)
x <- foreach::foreach(a=1:3, b=rep(10, 3)) %do% (a + b)  
```
At least that should reproduce the error according to https://github.com/rstudio/rstudio/issues/6692.

If yes, then I think we need to keep an eye on their thread. In the mean time you should be able to use GGIR by providing it with argument `do.parallel = FALSE` as that disables the whole attempt to use multiple cores.

angerhang, 2021-05-12:
> library("foreach")
> library("doParallel")
> cl <- parallel::makeCluster(1)
> doParallel::registerDoParallel(cl)
> x <- foreach::foreach(a=1:3, b=rep(10, 3)) %do% (a + b)

Ah thanks for the quick response again. It turns out that you are right, after updating the packages, everything seems to have worked fine :D 

There is no error using the commands you provided. Problem solved!

vincentvanhees, 2021-05-13:
That's great to hear Hang.
@RubenBrondeel I will close this issue now assuming this also resolves your issues. If you keep having problem feel free to re-open or start a new issue.

closed 2021-05-13 by vincentvanhees