wadpac/GGIR #409: subscript out of bounds

Schmay, 2021-03-29:
Hi 
GGIR shows an error message in part 5 when running the script on some outputs: 
Errors and warnings for 057267 ID 14.1.bin.RData$message
[1] "subscript out of bounds"

$call
`[<-`(`*tmp*`, di, fi:(fi + 2), value = c(ID, fnames.ms3[i], 
    wi))

I'm using the standard script from the homepage and have only changed the data and output directory as well as the maxdur (GGIR version 2.3-0).

How can I solve this problem?

Thanks

Schmay, 2021-03-30:
I think I found the issue. This error showed up on files where the participant did not wear the watch throughout the whole study period or the participant did not wear the watch during the day. I've reduced the maxdur on the first participant and there was no error. I'm still to figure out how to solve the problem with the latter participant.

vincentvanhees, 2021-03-30:
Thanks for reporting this, would it be possible for you to share the corresponding .RData file from the subfolder meta/basic produced by GGIR? My e-mail address is v.vanhees@accelting.com.

vincentvanhees, 2021-04-09:
I am closing this issue now. Feel free to re-open if you can share a more elaborate description to help reproduce the issue, and/or send me an example file.

closed 2021-04-09 by vincentvanhees

Schmay, 2021-04-11:
Good morning Vincent
Thank you very much for your message and sorry for the delayed response.
Unfortunately I was advised by the research team that I'm not allowed to
share the file because of ethical issues with this specific trial.
Nonetheless, I'm very grateful for your offer and will certainly send you
an example file if the error occurs again with a file that I am allowed to
share with you.
Thanks
Maya