wadpac/GGIR #412: Identify the separator in raw csv files

jhmigueles, 2021-04-09:
g.part1 does not work well with semicolon-separated csv's. A function to identify the separator of the raw csv's would be helpful to avoid data misreadings.

vincentvanhees, 2022-04-19:
Better option may be to transition to storing all 'csv' output as 'tsv' output.

We may want to do this gradually in phases:

2.7-0 (May 2022): Add it as optional arguments, and use this to test the functionality.
2.8-0 (September 2022): Add warning to csv users that csv will stop to be the default.
2.9-0 (December 2022): Change default to tsv, and warn users that  tsv is now the default

vincentvanhees, 2023-06-29:
Action plan:
- Start using fread also for raw acc files (ActiGraph and ad-hoc csv)

vincentvanhees, 2023-10-11:
closing this issue, because we already use fread everywhere

closed 2023-10-11 by vincentvanhees