wadpac/GGIR #419: g.applymetrics fails to calculate ENMOa

jhmigueles, 2021-05-03:
**Describe the bug**
ENMOa is not calculated, which produces that the M$metashort data frame gets empty and the next steps in GGIR do not work.

**To Reproduce**
Steps to reproduce the behavior.

1. Sensor brand: Not relevant (in my case, ActiGraph GT3X+)
2. Data format: csv
3. Approximate recording duration: 7 days
4. Are you using a sleep diary to guide the sleep detection: Not relevant, issue found in g.part1
5. Copy of R command used: 
g.shell.GGIR(mode = 1,          #Parts of GGIR to run
             #BASIC SETTINGS
             # datadir="C:/Users/chechu/OneDrive - KI.SE/acc_cap/acc_h?st2020/28_Bonden/barn/csv_raw_test", #where the raw acc.csv data is stored 
             # outputdir="C:/Users/chechu/OneDrive - KI.SE/Dokument/100 Work/130 Preschool policy intervention/136 Data/test raw data/Bonden", #where the results should be stored
             datadir="D:/accelerometry/storage/CAP/", #where the raw acc.csv data is stored 
             outputdir="D:/accelerometry/processing/CAP study/", #where the results should be stored
             studyname = "CAP",             
             f0 = 1, f1 = 1,
             overwrite = T, 
             print.filename = T, printsummary=T,
             myfun = myfun,
             #G.PART1: GET ENMO VALUES
             windowsizes = c(5,900,3600),     #First number is the epoch length
             desiredtz = "Europe/Stockholm",  
             do.enmo = T, do.enmoa = T, do.anglez = T,  #Metrics to calculate
             dayborder = 0,  
             #G.PART2: FIRST ESTIMATIONS OF PHYSICAL ACTIVITY 
             idloc = 2, 
             strategy = 1,
             hrs.del.start = 0, hrs.del.end = 0,
             maxdur = 0,
             includedaycrit = 16,    
             winhr = c(5,10),
             qwindow = c(6,23), 
             qlevels = c(960/1440,              #M1/3 (8h)
                         1320/1440, 1380/1440,  #M120, M60
                         1410/1440, 1430/1440,  #M30, M10
                         1435/1440, 1438/1440), #M5, M2
             ilevels = seq(0,800, by = 50),
             mvpathreshold = (8.6/85.7) * 1000,
             boutcriter = 0.8, bout.metric = 4, 
             #G.PART3-4: SLEEP CLASSIFICATION
             acc.metric = "ENMO",
             anglethreshold = 5,timethreshold = 5,
             def.noc.sleep=1,
             do.visual = FALSE, 
             outliers.only = FALSE,
             includenightcrit = 2,
             #G.PART5: MERGE SLEEP AND PHYSICAL ACTIVITY DATA (DEFINITIVE DATASET)
             threshold.lig = (5.3/85.7) * 1000, threshold.mod = (8.6/85.7) * 1000, threshold.vig = (15/85.7) * 1000,        
             boutdur.mvpa = c(1,5,10), boutdur.in = c(10,20,30), boutdur.lig = c(1,5,10), 
             boutcriter.mvpa = 0.8, boutcriter.in = 0.9, boutcriter.lig = 0.8, 
             timewindow = c("MM", "WW"),
             save_ms5rawlevels = TRUE, save_ms5raw_without_invalid = FALSE,
             save_ms5raw_format = "RData",
             data_cleaning_file=c(),
             #REPORTS
             do.report = c(),
             visualreport = F, 
             do.parallel = F)

6. Have you tried processing your data based on GGIR's default argument values? Does the issue you report still appear? YES, it still appears

**Expected behavior**
The M$metashort data frame should be correctly calculated with all the timestamps and metrics specified.
I'll set up a branch and fix this right away. The issue is in the line #217 in the function g.applymetrics. The code looks for negatives in the absolute values of ENMO. As they are inexistent, the vector is of length 0 and the whole data frame gets erased. Leaving out this line of code should solve the problem.

**Additional context**
Add any other context about the problem here.

**Before submitting**
- [X] Have you tried the steps to reproduce? Do they include all relevant data and configuration? Does the issue you report still appear there?
- [X] Have you tried this on the latest `master` branch from GitHub?

vincentvanhees, 2021-05-03:
Thanks Jairo. Yes, this sounds like a good solution.

jhmigueles, 2021-05-03:
I did it and I'm still getting the M$metashort data frame empty in part 1. I'm debugging the g.getmeta function but haven't found anything strange yet. Any idea about where could the probem be?

jhmigueles, 2021-05-04:
I found the problem in g.getmeta, see pull request

vincentvanhees, 2021-05-11:
Thanks Jairo. PR is now merged, sorry forgot tag this issue.
I will now close this issue.