wadpac/GGIR #423: CRAN request to remove closeAllConnections

vincentvanhees, 2021-05-11:
Just received this message from CRAN:

> Dear maintainers,
> 
> This concerns the CRAN packages
> 
>   Arothron BootstrapQTL CARRoT CENFA CNVScope CohortPlat DIZutils
>   EmiStatR GGIR Kernelheaping Luminescence OligoSpecificitySystem
>   PQLseq ProFound RAMClustR RForcecom Relexon SPUTNIK VineCopula
>   bRacatus chillR clam dCovTS distdrawr earlygating elasticsearchr
>   expands gfer gmvarkit hydroscoper insect kdevine learnr orddom osrmr
>   parallelSVM ptm rFIA rPanglaoDB rbacon rfml rmcfs robustSingleCell
>   rplum seqinr soilcarbon sybil tidyhydat uGMAR
> 
> maintained by one of you:
> ....
> 
> These package have code calling
> 
>   closeAllConnections()
> 
> which seems a very bad idea in package code, as indeed it will close all
> user connections etc, and not just the ones opened by the package.
> 
> Can you please fix "as necessary", and ensure your package code only
> closes connections it also opened?
> 
> Please fix before 2021-06-08 to safely retain your package on CRAN.
> 
> Best
> -k

vincentvanhees, 2021-06-01:
Now addressed, and 2.4-0 release submitted to CRAN.

closed 2021-06-01 by vincentvanhees

qy1028, 2024-11-26:
Hello, thank you for your excellent work for the GGIR package. I'm now analyzing the accelerometer data with GGIR (version 2.0-0). However, when running the function g.shell.GGIR, I recieved a warning message "invalid connection... Please make sure not to call closeAllConnections()." Is there any parameter I could set to avoid closing all connections with this version of GGIR? 
I appreciate for this valueable package and your help. Thank you!

qy1028, 2024-11-26:
> Now addressed, and 2.4-0 release submitted to CRAN.

Hello, thank you for your excellent work for the GGIR package. I'm now analyzing the accelerometer data with GGIR (version 2.0-0). However, when running the function g.shell.GGIR, I recieved a warning message "invalid connection... Please make sure not to call closeAllConnections()." Is there any parameter I could set to avoid closing all connections with this version of GGIR?
I appreciate for this valueable package and your help. Thank you!

vincentvanhees, 2024-11-26:
In general, I would not recommend using (very) old GGIR releases, especially the 2.0-0 release from 2020 was problematic. GGIR as research software has evolved a lot in the past 5 years.

However, if you insist on using 2.0-0 then in the [changelog](https://wadpac.github.io/GGIR/news/index.html#changes-in-ggir-version-24-0) you can see that closeAllConnections() was removed in 2.4-0 specially because newer R versions did not allow for it anymore. So, if you want to use an older GGIR version you also need to use the R version and possibly other GGIR dependencies that were available at the time.

Sorry, I do not have the resources to keep facilitating old releases. I only focus on issues and questions in the latest release. Help to maintain GGIR as open-source project is always welcome.