wadpac/GGIR #424: Planning for CRAN 4.3-0 release

vincentvanhees, 2021-05-11:
Given #423, I would like to make a new CRAN release with the following planning.

**May 10-21:**
- [x] Add fixes to issue #423 to PR #422
- [ ] Wait for feedback on PR #422 from @jhmigueles and/or @m-patterson. I hope at least one of you has time to do this before 21st of May?

**May 21-23:**
- [ ] Merge #422 in Master branch
- [ ] Create GitHub-only release version 4.2-3
- [ ] E-mail GGIR list to ask users to try out the version 4.2-3 and report urgent issues.

**May 24-31:**
- [ ] Wait for urgent issues and fix if any.

**June 1-3:**
- [ ] Prepare and submit 4.3-0 release for CRAN

vincentvanhees, 2021-05-11:
... I got the version numbers wrong, now corrected.

jhmigueles, 2021-05-13:
Hi Vincent,

I’ve been working on a tiny bug related to g.report.part2 this morning (see https://github.com/wadpac/GGIR/issues/425 and the branch https://github.com/wadpac/GGIR/tree/issue425_reportpart2stops). I’m not sure if I should do the pull request now or wait until you merge your branch on the fragmentation metrics with the master. I think we can include this one in the new release as well, what do you think?

vincentvanhees, 2021-05-13:
Hi Jairo, Feel free to make a separate pull request, we can merge them, no problem.
Yes, it would be good to include in this in the new release. Thanks

vincentvanhees, 2021-06-01:
I have now submitted 2.4-0 to CRAN, once it has been accepted by CRAN later this week I will also create 2.4-0 release here on GitHub.

closed 2021-06-01 by vincentvanhees