wadpac/GGIR #433: fix error in visual report generation following modification

ThomasKraft, 2021-06-10:
I am applying a minor custom modification to GGIR part1 output as previously suggested by @vincentvanhees. In brief, the code simply takes an input lookup table and uses timestamps at the beginning and end of wear time and uses those as cutoff points to set anything before or after to 'non-wear' time. This is an alternative to the 'strategy' options available:

```
specify_start_end <- function(folder_path, data_fn, deployment_file, start_time_buffer, end_time_buffer, desiredtz){

#  Load the milestone (RData) files generated by GGIR part 1 as stored in the meta/basic folder.
load(data_fn)
  
# Create an extra timestamp column in object M$metalong by converting the timestamps to POSIX value with GGIR function iso8601chartime2POSIX
M$metalong$timestamp_temp <- iso8601chartime2POSIX(M$metalong$timestamp, tz = desiredtz)  

# Change the non-wear score column in object M$metalong to 3 for all values outside your time interval of interest.
# load 
ofn <- gsub(pattern="(.*meta_)|(.[^\\.]*$)", replacement= "", data_fn) # parse the file name of the meta file to get the original data file name
st <- deploy$timestamp_start[match(ofn, deploy$filename)]
et <- deploy$timestamp_end[match(ofn, deploy$filename)]

# add in the time buffers as non-wear time
st <- st + start_time_buffer
et <- et - end_time_buffer

M$metalong$nonwearscore[which(M$metalong$timestamp_temp <= st)] <- 3  # set times before deployment to non-wear (3) 
M$metalong$nonwearscore[which(M$metalong$timestamp_temp >= et)] <- 3  # set times after end of wear to non-wear (3) 

# After you have done this, it is best to remove the new timestamp column, just in case the rest of the GGIR gets confused about seeing more columns than expected
M$metalong <- M$metalong[ ,-which(names(M$metalong) == "timestamp_temp")]

# - Save all object you found in the milestone file back to the same file, including the now modified object M.
# - Continue with part 2 with default settings for strategy. All time outside your value of interest will be treated as accelerometer non-wear time.
save(I, C, M, filefoldername, filename_dir, file= data_fn)
print(paste("Start and end times set at", st, "and", et, "respectively for file", data_fn, sep=" "))
rm(I, C, M, filefoldername, filename_dir)  # remove files from workspace
}
```

After applying this code, however, an error is thrown when generating visual reports, because the dataframe `sib.cla.sum` has 0 rows, resulting in "Error in 1:max(unique(S$night)) : result would be too long a vector
In addition: Warning message:
In max(unique(S$night)) : no non-missing arguments to max; returning -Inf" from `g.plot5.R`. 

The error originates because of the call at this line: https://github.com/cran/GGIR/blob/7f4617710db20539605b2b80b2d39d76c5a26934/R/g.plot5.R#L305. I believe this could be fixed by simply skipping the associated for loop with an if statement, eg. `if(nrow(S) >0 )`

I also wonder if any others would find it useful to have the function above built in to offer an alternative strategy based on start and end timestamps for wear time?

ThomasKraft, 2021-06-10:
Ah sorry, I need to do more tests first to make sure this issue isn't driven by other problems with my input files. My apologies!

vincentvanhees, 2021-06-18:
No problem Thomas. Please note that, as of last summer, GGIR has the option to take a diary file as input value for argument `qwindow`. For details see documentation for qwindow here: https://cran.r-project.org/web/packages/GGIR/GGIR.pdf

The diary file would then look along the lines of:
ParticipantID | Date | Start | End | Date | Start | End | Date | Start | End | Date | Start | End | Date | Start | End | Date | Start | End
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
YOURID001 | 07-05-18 | 0:00 | 24:00 | 08-05-18 | 0:00 | 24:00 | 09-05-18 | 0:00 | 24:00 | 10-05-18 | 0:00 | 24:00 | 11-05-18 | 0:00 | 24:00 | &nbsp; | &nbsp; | &nbsp;
YOURID014 | 10-05-18 | 0:00 | 24:00 | 11-05-18 | 0:00 | 24:00 | 14-05-18 | 0:00 | 24:00 | 15-05-18 | 0:00 | 24:00 | 16-05-18 | 0:00 | 24:00 | 17-05-18 | 0:00 | 24:00

I am not sure the 0:00 - 24:00 interval will work, as I did not design it with this use-case in mind. The aim was to select time segments shorter than 24 hours.

Let me know if you have trouble getting this to work.
Could this be a solution for what you want to do?

ThomasKraft, 2021-06-18:
Thanks @vincentvanhees for this helpful info! That certainly could work if the 0:00-24:00 interval doesn't cause problems. I'll give it a shot.

I will note, however, that the code I posted previously does seem to be working nicely. It should be relatively easy to integrate this with the existing functions given that it draws on input information that is already created, so another option would be for you to add this code to the repository as an additional "strategy" specification in which the user specifies `deployment_file`, `start_time_buffer`, and `end_time_buffer`. The code the cuts according to what essentially amounts to a diary with one row per person listing times that data collection started and ended (and the buffers specify how much additional time to ignore on either end).