wadpac/GGIR #455: g.part4 fails with "object 'undef' not found" error

l-k-, 2021-09-15:
**Describe the bug**
g.part4 fails on some datasets with the following error:

```
Error in rainbow(length(undef), start = 0.7, end = 1) : 
  object 'undef' not found
```

The error happens on [line 898](https://github.com/wadpac/GGIR/blob/d1dfd1674aa11bae96077410a69aba83de32ee5f/R/g.part4.R#L898) because variable `undef` is undefined. 

`undef` normally should get defined on [line 585](https://github.com/wadpac/GGIR/blob/d1dfd1674aa11bae96077410a69aba83de32ee5f/R/g.part4.R#L585) if the condition on [line 584](https://github.com/wadpac/GGIR/blob/d1dfd1674aa11bae96077410a69aba83de32ee5f/R/g.part4.R#L584) holds true. However, I have some data files where I end up with `calendar_date[j] == ""` and so this condition fails and `undef` never gets set.
(for this data file, I end up with `calendar_date == ["", "2/5/2017", "", "", "", "", "", "" ]`, and then at `j==6` the process crashes.

To be fair, the data files that are giving me trouble aren't actually good data -- it looks like the subject probably wasn't wearing the device much, if at all. So I get weird-looking results with one single 7-day sleep period across 8 days. So it's not useful data by any stretch. But I think even with not-so-nice data, GGIR should exit gracefully and not error out.

I think the solution should be similar to [this fix](https://github.com/wadpac/GGIR/commit/7adc45d2fcb392f5add725555ca03e662753dedc), and the [if (addlegend == TRUE)](https://github.com/wadpac/GGIR/blob/master/R/g.part4.R#L897-L906) block needs to be brought in one more level, inside the [if (nrow(spocum) > 1 & ncol(spocum) >= 5 & calendar_date[j] != "")](https://github.com/wadpac/GGIR/blob/d1dfd1674aa11bae96077410a69aba83de32ee5f/R/g.part4.R#L584) conditional block

**To Reproduce**
Steps to reproduce the behavior.

1. Sensor brand: Actigraph
2. Data format: gt3x file, parsed into a CSV using our own parser.
3. Approximate recording duration: 8 days
4. Are you using a sleep diary to guide the sleep detection: NO
5. Copy of R command used: 

`g.shell.GGIR(datadir="<...>/data", outputdir="<...>/results", do.report=1:5, visualreport=TRUE, do.visual=TRUE, do.enmo = FALSE, do.enmoa = TRUE, acc.metric="ENMOa", threshold.lig = c(61.84), threshold.mod = c(100.4), mvpathreshold =c(100.4), rmc.firstrow.acc = 2, rmc.col.acc = 2:4, rmc.col.time=1, rmc.unit.time = "UNIXsec", configtz = "UTC", desiredtz = "UTC", rmc.desiredtz = "UTC", rmc.sf = 80, rmc.check4timegaps = TRUE, rmc.doresample=FALSE, interpolationType=1)`

6. Have you tried processing your data based on GGIR's default argument values? Does the issue you report still appear? 

Not entirely. Since the data fed into GGIR is in a csv file, I had to specify some additional parameters, so that the csv file can be read in:

g.shell.GGIR(datadir="<...>/data", outputdir="<...>/results", do.report=1:5, visualreport=TRUE, do.visual=TRUE, rmc.firstrow.acc = 2, rmc.col.acc = 2:4, rmc.col.time=1, rmc.unit.time = "UNIXsec", rmc.sf = 80)

**Expected behavior**
I was expecting GGIR to exit gracefully, without throwing an error.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: MacOS 10.15.7
 - GGIR Version: 2.4-2

**Additional context**
I'm going to send a PR for the fix, since it's just one line, similar to [this fix](https://github.com/wadpac/GGIR/commit/7adc45d2fcb392f5add725555ca03e662753dedc)

**Before submitting**
- [] Have you tried the steps to reproduce? Do they include all relevant data and configuration? Does the issue you report still appear there? 
- [X] Have you tried this on the latest `master` branch from GitHub?

vincentvanhees, 2021-09-15:
Now closing this issue, see my comment in #456. Sorry about the miscommunication.

closed 2021-09-15 by vincentvanhees

l-k-, 2021-09-15:
@vincentvanhees I left a comment on [#456 ](https://github.com/wadpac/GGIR/pull/456#issuecomment-919961236) -- I think this issue still exists in your latest release. You had fixed a similar issue in the latest release, and I saw that, but your fix didn't go far enough.

vincentvanhees, 2021-09-15:
thanks @l-k- I agree, I have now merged your PR.

l-k-, 2021-09-15:
@vincentvanhees awesome, thank you for taking another look!