wadpac/GGIR #475: Error MainMetric not found

michelsara, 2021-11-30:
**Describe the bug**
When running the g.shell.GGIR() function on a csv file, it returns 
```
Error in `[.data.frame`(daysummary_tmp, , MainMetric) : 
  object 'MainMetric' not found
```
with the default values, moreover in the folder ms5.out doesn't appear anything.
With the bin file, it works with the default values, it gives the error when 
```
  strategy = 3,
  ndayswindow=1,
```

**To Reproduce**
Steps to reproduce the behavior.

1. Sensor brand: GENEActive
2. Data format: bin and csv
3. Approximate recording duration 14 days
4. Are you using a sleep diary to guide the sleep detection: NO
5. Copy of R command used: 
```
g.shell.GGIR(
  mode=c(1,2,3,4,5),
  datadir=datadir,
  outputdir=outputdir,
  overwrite = TRUE,
  f0=1, 
  f1=1,
  do.report=c(2,4,5),
  visualreport=TRUE,     
  dofirstpage = FALSE,
  viewingwindow=2)
``` 
launched with 
`source("~/Documents/GitHub/ActiSleep/GGIR.R", echo=TRUE)`

6. Have you tried processing your data based on GGIR's default argument values? Does the issue you report still appear? YES 

**Expected behavior**
The algorithm should finish without this error and do the visual report correctly.
**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS Monterey 12.0.1 
 - GGIR Version 2.5-0

**Additional context**
The problem didn't exist before updating from 2.4-3 to 2.5-0

**Before submitting**
- [x] Have you tried the steps to reproduce? Do they include all relevant data and configuration? Does the issue you report still appear there?
- [x] Have you tried this on the latest `master` branch from GitHub?

vincentvanhees, 2021-12-02:
I am not able to reproduce the issue with default values, but think I know where it goes wrong:

First of all, please note that removing a function argument from your g.shell.GGIR call, does not reset the argument value. You will have to either explicitly re-assign the argument value or delete the config.csv file in the output folder because that file is used as back-up if an argument is not provided.

It seems your error originates from using: `strategy = 3, ndayswindow=1,`

as that let's GGIR treat the entire measurement as invalid except the 24 most active hours.
These 24 hours do not have to align with a specific calendar day. You may end up with two half days of data, which are then flagged as invalid data and excluded. This then confuses the visualreport generation because it is looks for a column in the results that in normal situations would exist.

To fix this, you may could do:` includedaycrit = 8 `such that even days with only 8 hours are included. However, this may not work if one of your days has even less data.

If you want to identify and analyse the most active day in a recording then I think it is best to use default strategy=1, and then use the daysummary to identify the most active day yourself and go from there.

vincentvanhees, 2021-12-03:
Closing this issue now. Feel free to re-open if you feel this needs more discussion.

closed 2021-12-03 by vincentvanhees