wadpac/GGIR #478: read.myacc.csv function call in g.

jhmigueles, 2021-12-18:
**Describe the bug**
The function read.myacc.csv does not resample correctly the temperature when rmc.doresample = TRUE and the temperature sampling frequency is different (lower in my case) to the acceleration sensor sampling frequency.

**To Reproduce**
Steps to reproduce the behavior.

1. Sensor brand: Matrix
2. Data format: csv
3. Approximate recording duration 7 days
4. Are you using a sleep diary to guide the sleep detection: NO
5. Copy of R command used: 
     P = read.myacc.csv(rmc.file=files[2], 
                   rmc.dec=".", rmc.firstrow.acc = 1, 
                   rmc.skip = 0,
                   rmc.col.acc = 2:4, rmc.col.temp = 8, 
                   rmc.col.time=1, rmc.nrow=Inf,
                   rmc.unit.acc = "g", rmc.unit.temp = "C", 
                   rmc.unit.time = "POSIX",
                   rmc.format.time = "%Y-%m-%d %H:%M:%OS",
                   rmc.bitrate = c(), rmc.dynamic_range = 16, 
                   rmc.unsignedbit = FALSE,
                   rmc.desiredtz = "Europe/Madrid",
                   rmc.sf = 100,
                   rmc.headername.recordingid = "remarks",
                   rmc.check4timegaps = FALSE,
                   rmc.doresample=TRUE,
                   interpolationType=1)
6. Have you tried processing your data based on GGIR's default argument values? Does the issue you report still appear? YES
7.

**Expected behavior**
I expect having non-NA values on the temperature resampled data. All I get is NA's in the temperature column.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - GGIR Version 2.5-1

**Before submitting**
- [X] Have you tried the steps to reproduce? Do they include all relevant data and configuration? Does the issue you report still appear there?
- [X] Have you tried this on the latest `master` branch from GitHub?

vincentvanhees, 2021-12-22:
Thanks Jairo.
I just changed the label to feature, because this is not a bug of the existing functionality. The function resample based on the assumption that all signals are samples at the same rate. The function does not claim that it can handle unequal sample rates across signals.

vincentvanhees, 2023-12-06:
I am closing this now as it has been open for 2 years. I do not have a project where this is relevant. If this comes up again we can re-open it. As an intermediate solution users will have to interpolate the temperature data outside GGIR.

closed 2023-12-06 by vincentvanhees