wadpac/GGIR #514: Store GGIR version in all csv reports

vincentvanhees, 2022-02-16:
Storing GGIR version in all csv reports would ease keeping track of what version of GGIR was used.
In my own experience it can sometimes take months between running an analysis and reporting about it, a time frame which can lead to confusion about the specific version used.

vincentvanhees, 2022-02-22:
One more thought: The GGIR version is also stored in the config.csv file, so technically the user has a record of the GGIR version. However, I think it would be good to also have it in the results to reduce risk for confusion

An entirely different approach is that we store the config.csv filename with the specific time and date of the analysis, and also store that time and date in the csv reports. The problem with this is that it would require redesigning the whole configuration file process as there will be multiple configuration files, one for each iteration of the analysis.... So, I think the best approach ise to simply store the GGIR version in the csv reports and expect the user to keep track of what config.csv or R script they used.

jhmigueles, 2022-02-23:
What about making subfolders in results for each iteration of the analyses, so that the user know what GGIR version was used in the reports used and can go back to previous versions. Something like:
- output_mystudy/results/v2.6-0/
- output_mystudy/results/v2.6-1/

This way a new folder with reports would be generated every time the version of GGIR changes, which might be a good idea so that analyses from different versions are not mixed (e.g., results from part 4 and part5 being inconsistent because they were analysed with different versions).

vincentvanhees, 2022-02-23:
Thanks, I see advantages with your proposal, but also some disadvantages:
- It creates a false sense of comparability because different versions may have been used together with different input arguments. We would then also have to store the config.csv file in each of the results folders.
- It creates a lot of output folders, which some users may find frustrating as they also take up space. Further, the user will have to search the output folders to identify when GGIR was last run for a certain part. 
- Results are no longer in a predictable directory path. Having the results in a predictable directory path makes it easier for users to write R scripts for post-processing. For example, if the folders are version specific then a user may continue to look in folder v2.6-0 while forgetting to change their post-processing scripts to look at folder v2.6-1.

I now realise that we will have the same problems if we include the GGIR version in the .csv filenames. Adding GGIR version as a new column inside the .csv files will take up a lot of space. So, I guess it is not as easy as I thought it would be...

The advantage of the existing approach in GGIR is that we make the user responsible for keeping track of versions and consistency. This is easiest for us in terms of maintenance and development.

So, how about we close this issue? or can you see an alternative way around this?

vincentvanhees, 2022-02-24:
No, people can just open the file and look inside it, and we pay the price of yet another GGIR function to maintain. So, let's stick to the strategy that the user is responsible for archiving their results and config.csv files.

closed 2022-02-24 by vincentvanhees