wadpac/GGIR #518: Compatibility with GENEActiv .csv files

vincentvanhees, 2022-02-22:
**Problem:**
The GGIR documentation states that GGIR is compatible with the GENEActiv .csv file format, but I have not used this myself for a very a long time. I originally implemented compatibility with the .csv files, but I also know that the file format kept changing between GENEActiv software releases, which complicated sustaining the compatibility. Once we were able to work with the .bin files via the GENEAread package I lost interest in the .csv files.

If we keep claiming that GGIR is compatible with GENEActiv .csv we have two problems
- We do not know whether GGIR actually is compatibility any more.
- More data formats means more maintenance for GGIR.

Instead, it may be easier to tell users who only have GENEActiv data in .csv format to use GGIR's read.myacc.csv functionality instead. In this way we can slowly work towards a situation where read.myacc.csv is the preferred approach for handling .csv-file formats.

**Proposal:**
- Remove all claims about GENEActiv .csv compatibility from the documentation.
- Deprecate all GGIR functionality that is specific to the GENEActiv .csv file format.

@jhmigueles do agree with this approach?

jhmigueles, 2022-02-22:
I was thinking of the same thing for ActiGraph and axivity csv files. I have not tried to read GENEActiv csv files myself, but I was not successful with the axivity csv files. Should we be consistent and suggest the read.myacc.csv function for all these files?

For ActiGraph, maybe we should fist make sure that the reading of the gt3x files is working well, or we can set the read.myacc.csv arguments internally upon the detection of the ActiGraph brand (as we know how their csv's look like).

What do you think? Should we do the same for all brands?

vincentvanhees, 2022-02-22:
That would require us to carefully test that read.myacc.csv works with all those data formats, which could be a lot of work.

The ActiGraph file format is too odd to fit in the read.myacc.csv functionality, so I think it is easier to keep the ActiGraph csv format functionality. Also, because .gt3x has only recently been added and only works with newer .gt3x files. Maybe we can revisit this in a couple of years but for now I think it is best to leave it as it is.

For Axivity I did a project to make sure it works for one specific instance of the .csv data. So, I am hesitant to change that as those people may not like that the functionality is removed.

So, I prefer to take this step by step and only worry about GENEActiv csv format for now as I am almost certain this has never worked well. Next step could be to slowly spend more time on read.myacc.csv function, e.g. better documentation and more user friendly. Once that has matured it will be easier to at some point stop the functionality for Axivity csv.

jhmigueles, 2022-02-28:
Does this also apply to genea csv files? Or only to geneactiv? Never seen genea csv files but I assume they have not changed and GGIR is able to handle them.

vincentvanhees, 2022-02-28:
Genea devices were non-commercial and used in many of the initial studies. It has nothing to do with GENEActiv.
When GENEActiv started they entirely redesigned the data structure.

However, GENEA files also store .bin files although you need GGIR::g.binread to read them because the structure is different, and they also store .csv files..

 think it is best not to deprecate the .csv file functionality for Genea as there may be studies who still use those files and never stored the corresponding .bin files.

jhmigueles, 2022-02-28:
Great, I was not aware of the difference between genea and GENEActiv, thanks for explaining.

**Proposal**
1. Leave g.inspectfile function as it is, so that GGIR is still able to identify GENEActiv csv files (yet not to read them)
2. Call to the stop() function right after the g.inspectfile function is used in the rest of functions (if mon == 2, dformat == 2 & length(rmc.firstrow.acc) == 0)
3. Note that the processing would then stop and run an ERROR message. This could tell somethin like: the geneactiv csv reading functionaily is deprecated in GGIR v.2.6-2, please use either the bin files or the read.myacc.csv function
4. Remove all the code related to reading geneactiv csv files in g.readaccfile

@vincentvanhees are you ok with this?

vincentvanhees, 2022-04-06:
Once the user specifies argument `rmc.firstrow.acc` GGIR concludes [here](https://github.com/wadpac/GGIR/blob/master/R/g.inspectfile.R#L205-L207) that monitor type is 0 (other brand) and data format is 5 (ad-hoc .csv). Therefore, it should be impossible that a file that is intended to be run as ad-hoc .csv file is accidentally read as ActiGraph file, or that an ActiGraph file is accidentally processed as an ad-hoc csv file. So, I think that testing this with one ad-hoc .csv file format should be sufficient. No need to test a whole range of variations on ad-hoc .csv formats.

PR #543 (2022-04-13): Issue518 geneactiv csv

The GENEActiv csv reading function through g.readaccfile is now deprecated. GGIR stops and suggests the user to use either the bin files or the read.myacc.csv function. Also, axivity csv files can now be read after a couple of adaptations to g.getidfromheaderobject and g.imputeTimegaps.
Fixes #518 

Checklist before merging:

- [x] Existing tests still work (check by running the test suite, e.g. from RStudio).
- [x] Added tests (if you added functionality) or fixed existing test (if you fixed a bug).
- [x] Updated or expanded the documentation.
- [x] Updated release not [... cut]