wadpac/GGIR #522: Impute very long time gaps (idle sleep mode, ActiGraph)

jhmigueles, 2022-02-23:
**Is your feature request related to a problem? Please describe.**
When the idle sleep mode is used in ActiGraph devices, GGIR finds difficult to impute large time gaps in which the device has entered in sleep mode (several days/weeks). GGIR tries to impute all the missing data at once, which may trigger memory issues and stop running the data processing.

**Describe the solution you'd like**
Large time gaps should be imputed in smaller chunks to ensure that the memory can handle the data.

**Additional context**
This would only occure when processing the ".gt3x" files, when converted to ".csv" the files are processed with no issues.

jhmigueles, 2022-07-14:
Now GGIR prompts a preventive message suggesting the user to use csv files instead of gt3x in the case that the R session crashes (included in the branch issue573_imputeTimeGaps)

vincentvanhees, 2022-07-18:
I am not sure this is what we want. There is no danger for most datasets and as a result it unnecessarily gives the impression that GGIR is unstable.

Instead I propose to detect whether the total amount of time gaps per data chunk to be imputed sums to more than a 24 hours. If yes, give error and stop processing the file, including a message that the processing was stopped to prevent excessive memory usage. In that way we ensure that GGIR never crashes and always gives a meaningful error message.

jhmigueles, 2022-07-18:
I was working on this and I realized it was not crashing when imputing the time gaps but way before when reading the data in chunks. It seems that this was solved when merging this [PR](https://github.com/THLfi/read.gt3x/pull/40). I updated the read.gt3x package and now my test file gets processed without any problem.

I cannot find any file that makes my r session crash now, so maybe we should close this issue and only reopen it in the future if we get any other problematic file. Or we can suggest installing the read.gt3x version >= 1.2.0 in the case an older version is used. What do you think?

jhmigueles, 2022-08-20:
I'm a bit stuck here. I got to impute the data at epoch level in the branch "issue622_LongGaps" and metashort data looks good. However, the calculation of the nonwear time and clipping gets compromised. 

The data frame "data" with the raw data now still presents time gaps, since the imputation has been performed at epoch level. So, when defining the nonwear and clipping in g.getmeta, the code will take whichever number of samples fill the ws2 and ws windows, without considering that there might be time gaps in there (and then compromising the windows lengths).

The only solution I can think ok is re-imputing the raw data to match the window sizes with the number of samples. But, if I do so, we gain nothing from keeping the imputation for the epoch level. 

Any proposal??

vincentvanhees, 2022-09-07:
Closing this issue now, because fixes by #631

closed 2022-09-07 by vincentvanhees