wadpac/GGIR #533: Shell script fails with error message [1] "'from' must be a finite number"

AJDooley86, 2022-04-04:
Hi Everyone,

I have a very (very!) limited knowledge of coding, and use the GGIR shell function to easily process GENEActiv .bin files so that I can interpret physical activity and sleep metrics.

This has been working fine, but recently I've been getting an error that I haven't been able to troubleshoot. Apologies if this is a 'rookie mistake' or if it's been covered elsewhere, but hopefully someone can tell me what I'm doing wrong! Any help gratefully appreciated.

Shell I've been trying to run:

g.shell.GGIR(datadir 		 = 	"C:/Users/adamd/Desktop/ADTESTMARInputs",
             outputdir 		         =     "C:/Users/adamd/Desktop/ADTESTMAROutputs",
             desiredtz		         =	'Pacific/Auckland',
             loglocation                  = "C:/Users/adamd/Desktop/ADTESTMARInputs/SleepLog.csv",
             relyonsleeplog            = FALSE,
             sleeplogidnum            = FALSE,
             colid = 1,
             coln1 = 2,
             mode		=	c(1,2,3,4,5),
             strategy		=	2,
             maxdur		=	14,
             nnights		=	13,
             timewindow		=	'MM',
             overwrite		=	TRUE,
             visualreport	=	TRUE,
             do.visual		=	TRUE,
             do.report		=	c(2,5),)

and the output:

__________________________________________________________________________________________
 Part 1 

Checking that user has read access permission for all files in data directory: Yes
Checking that user has write access permission for directory specified by argument outputdir: Yes

 Busy processing ... see C:/Users/adamd/Desktop/ADTESTMAROutputs/output_ADTESTMARInputs/meta/basic for progress

Errors and warnings for 592 1.bin$message
[1] "'from' must be a finite number"

$call
seq.default(starttime3, (starttime3 + ((nrow(metashort) - 1) * 
    ws3)), by = ws3)

Errors and warnings for 592 2.bin$message
[1] "'from' must be a finite number"

$call
seq.default(starttime3, (starttime3 + ((nrow(metashort) - 1) * 
    ws3)), by = ws3)

__________________________________________________________________________________________
 Part 2 

 Busy processing ... see C:/Users/adamd/Desktop/ADTESTMAROutputs/output_ADTESTMARInputs/meta/ms2.out for progress

Errors and warnings for NA$message
[1] "cannot open the connection"

$call
readChar(con, 5L, useBytes = TRUE)

Errors and warnings for NA$message
[1] "cannot open the connection"

$call
readChar(con, 5L, useBytes = TRUE)

__________________________________________________________________________________________
 Part 3 

 Busy processing ... see C:/Users/adamd/Desktop/ADTESTMAROutputs/output_ADTESTMARInputs/meta/ms3.out for progress

Errors and warnings for NA$message
[1] "cannot open the connection"

$call
readChar(con, 5L, useBytes = TRUE)

Errors and warnings for NA$message
[1] "cannot open the connection"

$call
readChar(con, 5L, useBytes = TRUE)

__________________________________________________________________________________________
 Part 4 
Error in `[.data.frame`(S, , coln1 + ((i - 1) * 2)) : 
  undefined columns selected

AJDooley86, 2022-04-04:
Hi Vincent,

Thanks for your response. to answer your questions;

1. I am using the GENEActiv Original from ActivInsights.
2. I'm working with .bin files, but have tried with .csv as well and keep running into the same issue.
3. Yes, in the preview window of the GENEActiv software there is data, and the file size indicates that it collected for a few days.
4. It's only affecting files I've produced recently. When I run a previous file through the same shell function (.bin file created last August) it processes without a hitch and produces the outputs. Since February I've been getting this error. I've tried with two files, both collected from different watches and have the same issue. It gets stuck generating the 'basic' meta file, and thus can't progress to the other steps. I've tried to run it on a different computer as well, but same problem. I'm pretty sure I've been using the latest version of RStudio and GGIR.

Thanks again for looking into it, really appreciate your help.

Adam

vincentvanhees, 2022-04-07:
Can you try:

```
library(GENEAread)
data = GENEAread::read.bin(binfile = "C:/yourfile.bin", start = 1, end = 100)
print(head(data$data))
print(head(data$header))
```

If this works then we at least know that the data can be read. If not then the problem seems to be specific to the bin files or their permissions.

If the above works then try to process the file again with g.shell.GGIR argument `do.parallel = FALSE` as that would provide a bit more console output to help investigate the issue.

AJDooley86, 2022-04-08:
Cool - GENEAread works, there does appear to be data there and it can be read. Rerun of GGIR produces the following:

g.shell.GGIR(datadir 		= 	"C:/Users/adamd/Desktop/ADTESTMARInputs",
             outputdir 		=	"C:/Users/adamd/Desktop/ADTESTMAROutputs",
             desiredtz		=	'Pacific/Auckland',
             loglocation = "C:/Users/adamd/Desktop/ADTESTMARInputs/SleepLog.csv",
             do.parallel = FALSE,
             relyonsleeplog = FALSE,
             sleeplogidnum = FALSE,
             colid = 1,
             coln1 = 2,
             mode		=	c(1,2,3,4,5),
             strategy		=	2,
             maxdur		=	14,
             nnights		=	13,
             timewindow		=	'MM',
             overwrite		=	TRUE,
             visualreport	=	TRUE,
             do.visual		=	TRUE,
             do.report		=	c(2,5),)

GGIR version:  2.6-0

   Do not forget to cite GGIR in your publications via a version number and
   Migueles et al. 2019 JMPB. doi: 10.1123/jmpb.2018-0063. 
   See also: https://cran.r-project.org/package=GGIR/vignettes/GGIR.html#citing-ggir

   To make your research reproducible and interpretable always report:
     (1) Accelerometer brand and product name
     (2) How you configured the accelerometer
     (3) Study protocol and wear instructions given to the participants
     (4) GGIR version
     (5) How GGIR was used: Share the config.csv file or your R script.
     (6) How you post-processed / cleaned GGIR output
     (7) How reported outcomes relate to the specific variable names in GGIR
__________________________________________________________________________________________
 Part 1 

Checking that user has read access permission for all files in data directory: Yes
Checking that user has write access permission for directory specified by argument outputdir: Yes

P1 file 1

Investigate calibration of the sensors with function g.calibrate:
Number of pages in binary file: 87080 
Processing...
==========================================================================================
Processing took: 0.007 secs .
Loaded 900 records (Approx  0 MB of RAM)
22-02-21 20:30:00.500 (Mon)  to  22-02-21 20:30:22.974 (Mon) 

Loading chunk: 1Number of pages in binary file: 87080 
Processing...
==========================================================================================
Processing took: 6.438 secs .
Loaded 1741800 records (Approx  98 MB of RAM)
22-02-21 20:30:00.500 (Mon)  to  22-02-22 08:35:45.474 (Tue) 
 2Number of pages in binary file: 87080 
Processing...
==========================================================================================
Processing took: 6.021 secs .
Loaded 1741800 records (Approx  98 MB of RAM)
22-02-22 08:35:45.500 (Tue)  to  22-02-22 20:41:30.474 (Tue) 
 3Number of pages in binary file: 87080 
Processing...
==========================================================================================
Processing took: 5.921 secs .
Loaded 1741800 records (Approx  98 MB of RAM)
22-02-22 20:41:30.500 (Tue)  to  22-02-23 08:47:15.474 (Wed) 
 4Number of pages in binary file: 87080 
[... 75 lines cut ...]
22-02-26 21:26:53.000 (Sat)  to  22-02-27 21:38:15.474 (Sun) 
 7Number of pages in binary file: 87080 
Splitting into  2  chunks.
Processing...
==========================================================================================
Processing took: 12.2 secs .
Loaded 3483300 records (Approx  195 MB of RAM)
22-02-27 21:38:15.500 (Sun)  to  22-02-28 21:49:37.974 (Mon) 
 8Number of pages in binary file: 87080 
Processing...
==========================================================================================
Processing took: 6.624 secs .
Loaded 1740900 records (Approx  97 MB of RAM)
22-02-28 21:49:38.000 (Mon)  to  22-03-01 09:55:00.474 (Tue) 
Error in seq.default(starttime3, (starttime3 + ((nrow(metashort) - 1) *  : 
  'from' must be a finite number
In addition: There were 50 or more warnings (use warnings() to see the first 50)

--

I note that when the g.getmeta function runs it returns "date not recognized".

This has jogged my memory. A year or so ago we were having trouble processing files and my colleague had determined that there was some problem with the 'Pacific/Auckland' timezone - specifically with respect to our daylight savings shift. It seemed that files produced at a certain time of the year would fail to run, but files from the other six months would work fine. We never got to the bottom of the issue but the workaround was to use a virtual machine running R in Ubuntu and this worked, so we assumed the problem was specific to Windows in some way. Do you think this could be the same issue? 

I'm away and don't have access to my virtual machine right now, but can try to see if it will process these files in a week or so when I get back.

AJDooley86, 2022-04-08:
These are the some of the warnings:

Warning messages:
1: In readLines(fc2, n = headlines) : line 26 appears to contain an embedded nul
2: In readLines(fc2, n = headlines) : line 27 appears to contain an embedded nul
3: In readLines(fc2, n = headlines) : line 28 appears to contain an embedded nul
4: In readLines(fc2, n = headlines) : line 29 appears to contain an embedded nul
5: In readLines(fc2, n = headlines) : line 30 appears to contain an embedded nul
6: In readLines(fc2, n = headlines) : line 32 appears to contain an embedded nul
7: In readLines(fc2, n = headlines) : line 45 appears to contain an embedded nul
8: In readLines(fc2, n = headlines) : line 26 appears to contain an embedded nul
9: In readLines(fc2, n = headlines) : line 27 appears to contain an embedded nul
10: In readLines(fc2, n = headlines) : line 28 appears to contain an embedded nul

vincentvanhees, 2022-04-08:
Are you able to share the problematic file with me (v.vanhees@accelting.com)? I can have a look whether I can reproduce the issue.

Regarding...
```
Warning messages:
1: In readLines(fc2, n = headlines) : line 26 appears to contain an embedded nul
2: In readLines(fc2, n = headlines) : line 27 appears to contain an embedded nul
3: In readLines(fc2, n = headlines) : line 28 appears to contain an embedded nul
```
this is R package GENEAread specific and not related to GGIR. I tend to ignore these warnings, and believe I reported it to maintainers at some point via email as they do not have public repo.

vincentvanhees, 2022-04-18:
I am able to process your file with latest GitHub version of GGIR:

```
library(remotes)
remotes::install_github("wadpac/GGIR")
```

Further, I had to add arguments:
`idloc = 6`
and I had to change the first column in the sleeplog to start with:

ID  |  onset_N1
------| -------
592 1 |  22:09:00
592 2 | 22:20:00

Because ID needs to match ID in part2_summary.csv file.

It looks like that in your previous GENEActiv data the ID was conveniently stored in the GENEActiv file header while in the new GENEActiv data you did not enter the ID when configuring the accelerometers, by which GGIR can understandably not extract it. This is why you will now need to use argument `idloc` in order to extract the ID from the filename.

closed 2022-04-18 by vincentvanhees

charleslucenap, 2023-11-01:
Hello all,

I'm with the same problem as well. I've got a database with data from 1014 participants. However, after part 1, the R program presents the following: "$call Ops.data.frame(meta_temp[1:(nrow(meta_temp) - 1), 2:7], meta_temp[2:nrow(meta_temp),  2:7])"

After that, in part 4, the R program generates just 860 files instead of 1014 (original). Probably, I think that these differences are about this "error". 

In part 5, I received the following message as well: "Error in 1:max(summarysleep_tmp$night) : "

I am using the GENEActiv, as well as .bin file. Curiously, these data were performed with other cut-offs in the past, and everything works well. However, I'm updating the cut-off and identified these errors.

Could you help me, please?

Charles.

vincentvanhees, 2023-11-01:
The discussion above indicates the mistake the others made with `idloc` and `colid`. You do not clarify whether that could also be the case for you and you do not share details on how you ran GGIR and what version you used. Please start a new issue with those details included.