wadpac/GGIR #535: Restrict `Ncores2use` to the length of the outputlist

tarensanders, 2022-04-06:
**Is your feature request related to a problem? Please describe.**
I'm in the process of reprocessing tens-of-thousands of accelerometer files from multiple previous studies for a project. I have access to a high-performance machine (112 cores, 196GB RAM), and I am splitting the work up on a per-study basis (to make meta-data matching simpler later on). In some cases, the number of files to be processed for a study are less than the number of cores I have available.

When you process files in parallel, GGIR creates a cluster equal either to the number of cores available on the system less one, or the number of cores that the user specifies (whichever is lower). 

https://github.com/wadpac/GGIR/blob/8b9c83fbfc775f52e68aa42c99bc202c79439312/R/g.part1.R#L419-L420

and

https://github.com/wadpac/GGIR/blob/8b9c83fbfc775f52e68aa42c99bc202c79439312/R/g.part1.R#L441-L444

In instances where this is less than the number of files to process, this results in workers being created that just sit idle. For example, the machine is currently doing a study with only 35 participants, but it's created 111 workers. This isn't really a big deal, but it does create a bunch of unnecessary overhead in both creating and destroying the cluster. Each worker also seems to take up ~750Mb of memory.

**Describe the solution you'd like**
 I could solve this by getting the number of files to processes and passing it to `maxNcores`. But for other users, it might be better for GGIR to manage this? GGIR already knows the number of files to process because it has the variable `f1`:

https://github.com/wadpac/GGIR/blob/8b9c83fbfc775f52e68aa42c99bc202c79439312/R/g.part1.R#L135-L140

So a simple solution would be to revise

https://github.com/wadpac/GGIR/blob/8b9c83fbfc775f52e68aa42c99bc202c79439312/R/g.part1.R#L442

to

```R
Ncores2use = min(c(Ncores - 1, params_general[["maxNcores"]], f1))
```

I'd be happy to open a PR with this change, if you think it's worth implementing.

vincentvanhees, 2022-04-07:
Great idea @tarensanders. If you could PR that would be much appreciated.

I propose implementing it as:
`Ncores2use = min(c(Ncores - 1, params_general[["maxNcores"]], (f1 - f0) + 1 ))`

to make sure it also works for the use case where f0 is not 1. For example, when processing files on a computing cluster it is good to process files in batches. You would then write a bash script that specifies which file range (f0 to f1) should be processed per cluster job.

Would you mind repeating this update for:

g.part2:

https://github.com/wadpac/GGIR/blob/8b9c83fbfc775f52e68aa42c99bc202c79439312/R/g.part2.R#L209

g.part3:

https://github.com/wadpac/GGIR/blob/8b9c83fbfc775f52e68aa42c99bc202c79439312/R/g.part3.R#L127

g.part5:

https://github.com/wadpac/GGIR/blob/8b9c83fbfc775f52e68aa42c99bc202c79439312/R/g.part5.R#L969

tarensanders, 2022-04-07:
Good point about batching. I'll open the PR now.

commit 28d7ab76 (2022-04-08): Restrict `Ncores2use` to the number of files
Fixes #535
closed 2022-04-08 by vincentvanhees