wadpac/GGIR #545: Annotated time-series with activity labels

gkiar, 2022-04-15:
**Is your feature request related to a problem? Please describe.**
When using GGIR, we can get summary PDF reports that include annotations for the activity-type at given moments in time. However, it is unclear how to extract such an annotated timeseries with activity and sleep labels so that the data may be used directly.

**Describe the solution you'd like**
A possible output from the GGIR 5-part pipeline that includes the annotated timeseries data saved to a PDF. The annotated timeseries file would, ideally, include columns for: timestamp, x accel, y accel, z accel, enmo, zangle, <binary labels for each activity, rest/sleep, and wear state)

**Describe alternatives you've considered**
There isn't really much of an alternative, other than re-implementing the algorithms used within GGIR or using an alternate tool for processing, as at the end of the day my analysis requires an annotated timeseries.

Thanks for your consideration!

vincentvanhees, 2022-04-18:
I think GGIR already offers this, see GGIR documentation for arguments:

- **save_ms5rawlevels** Boolean, whether to save the time series classification (levels) as a csv files.
- **save_ms5raw_format** Character string to specify how data should be stored: either "csv" (default) or "RData". Only used if save_ms5rawlevels=TRUE.
- **save_ms5raw_without_invalid** Boolean to indicate whether to remove invalid days from the time series output files. Only used if save_ms5rawlevels=TRUE.

This will include magnitude of acceleration and the classifications of intensity levels (both bouted and unbouted), and sleep. If you also want additional acceleration metrics then export those with argument **epochvalues2csv** and merge the two time series.

gkiar, 2022-04-18:
Excellent, thanks so much, this worked well! Apologies for my negligence, I'm rather new to using the tool.

I may have follow-up questions on how to parse each of the class labels, but will refer to the documentation, first.

Thank you again,
Greg

gkiar, 2022-04-18:
Quick follow-up question:

When looking at waking activity during the sleep window (e.g., `spt_wake_VIG`), I'm assuming this is unbouted. Is that correct?

vincentvanhees, 2022-04-19:
Yes, it is unbouted.