wadpac/GGIR #550: Rename g.shell.GGIR?

vincentvanhees, 2022-04-20:
Given that `g.shell.GGIR` is hard to pronounce, I am wondering whether we should rename it.

For example:
1. shell => attempt to be consistent with current name, but just shorter
2. pipeline => to emphasize that GGIR is like a pipeline
3. GGIR => attempt to be consistent with current name, but just shorter. It also emphasizes that this is the central function of GGIR.

We can still keep `g.shell.GGIR` as a separate function, but use it as a wrapper around the renamed function. In that way we do not break consistency and GGIR users can continue using their existing R scripts.

I welcome feedback or suggestions for alternative names.

arcticFox-git, 2022-04-29:
Hi Vincent,

I would agree with the last paragraph regarding the wrapper function and consistency. While it may seem like a small change, some researchers may have complex setups and even going through and renaming the function in their script would introduce unnecessary work. Especially if they aren't active on the repo or google group, even finding out why their script is no longer functioning may be an issue.

In terms of new names, I would pass my vote for GGIR. This makes sense given that the shell function is the entirety of the process.

tarensanders, 2022-05-10:
I'd also vote for `GGIR::GGIR()` as the 'main' function.

closed 2022-05-24 by vincentvanhees