wadpac/GGIR #551: Argument configtz and desiredtz not well used for .gt3x data

vincentvanhees, 2022-05-02:
It seems that argument configtz and desiredtz are not used when reading .gt3x data. The code seems to default to local timezone.
I am now investigating and proposing a solution.

m-patterson-wearable, 2022-05-31:
I ran into a similar issue when I exported data using Centre-Point 3. It creates .gt3x files with timezone UTC, then states in the header what the timezone offest is. I created a fix, but I think it would only apply to using .gt3x files that are exported out of Centre-Point, I wouldn't want to break functionality for .gt3x files that come from elsewhere.

The problem for batch processing is that one desiredtz / configtz might not work for files that come from different timezones.

This is the code, starting a line 175 in g.getstarttime.R:

`starttime = as.POSIXct(as.character(P[1, 1]), tz = "UTC") # gt3x CP exports in GMT
 tz_offset = as.integer(strsplit(toString(header["TimeZone", ]), ":")[[1]][1])
 starttime = starttime + (tz_offset * 60 * 60)`

angerhang, 2022-06-13:
Yes, if my understanding is right. https://github.com/actigraph/GT3X-File-Format says that the `timestamp` variable is
> The date and time of the data contained in the record are marked to the nearest second in [Unix time](http://en.wikipedia.org/wiki/Unix_time) format.

When I processed a file initialised using ActiLife with non-UTC time zone, the `timestamp` stored is always UTC, the true Unix time. However the header file doesn't factor in day light saving. I had one file in the winter and another one in the summer. Both of them had the same time shift. 

A better way to do this correctly is to prompt the user to enter the time zone information so that we can parse `timestamp` and day light saving properly.

vincentvanhees, 2022-08-10:
I have now looked into it and think I fixed it.

Instead of extracting a timezone from the file header that may or may not account for DST, I do the following.

In line with other GGIR functions I allow user to specify arguments:
desiredtz => timezone where accelerometer is worn, defaults to timezone of system
configtz => timezone where accelerometer is configured, default NULL

Next, I only convert the first timestamp in the data, and use this to extrapolate to other epochs in the data based on known epoch length. In the code below `P[1,1]` is the first timestamp in the data. I am doing this, because we need to convert the timestamp to character format which would be very compute intensive if we did it for the raw data. 

**Code (inside g.getstarttime):**

```
if (is.null(configtz)) configtz = desiredtz
starttime = as.POSIXct(as.character(P[1, 1]), tz = configtz)
if (configtz != desiredtz) {
  attr(starttime, "tzone") <- desiredtz
}
starttime = as.POSIXlt(starttime, tz = desiredtz)
```

@angerhang - sorry, these are all R specific functions, but I hope you can follow the logic: I interpret the timestamp as character to ignore timezone information, I then assign configtz, and finally convert to desiredtz and POSIXlt.

I will integrate this update in new pull request later today, and then merge it in GGIR master branch.

PR #611 (2022-08-10): Issue599 upgrade readgt3x

Fixes #599, which replaces a lot of code by a direct dependency on read.gt3x.
Fixes #551, which modifies how starttime is extracted.

I have tested this with the problematic file mentioned in PR #557 in Windows and in Ubuntu, and results are correct now.

Checklist before merging:

- [x] Existing tests still work (check by running the test suite, e.g. from RStudio).
- [ ] Added tests (if you added functionality) or fixed existing test (if you fixed a bug).
- [ ] Updated or expanded the documentation.
- [x] Updated release notes in `inst/NEWS.Rd` with a user-readable  [... cut]
closed 2022-08-10 by vincentvanhees