wadpac/GGIR #556: Clarification on .bin file size

jmkurka83, 2022-05-06:
I am processing a .bin file that is ~430 mb, but when GGIR finishes, the part 3 sib.cla.sum dataset only shows 1 night of data and the daysummary indicates 11.5 hours. Report part 4 tries to print, but indicates there is no data. The visual reports try to print, but there is an error indicating object MainMetric is not found (I'm assuming it doesn't exist due to lack of data).

This .bin file is the same size as others I've processed that include 10+ nights of sleep. Could there be a reason GGIR isn't picking up the additional data?

Thank you!

vincentvanhees, 2022-05-11:
I would be surprised if file size had anything to do with the issue. A few question that may help pinpoint the cause of the issue:
- Do I understand correctly that part 1 and 2 ran without errors or warnings?
- Are you getting this issue on an older laptop with < 4 GB of RAM memory? If yes, did you try argument chunksize= 0.3 for example?
- Is sample frequency correctly recognised in part2_summary.csv and in line with your expectations?
- Does the pdf file in the QC folder also indicate that there is only 1 day or so of data?
- Have you tried processing on a different machine?
- Which version of GGIR are you using?

vincentvanhees, 2022-05-27:
Closing this issue now, feel free to re-open if you problem persists.

closed 2022-05-27 by vincentvanhees

jmkurka83, 2022-06-21:
Hi Vincent! Sorry for the delay in following up. I'm using GGIR 2.7-1 and unfortunately I don't have a different machine I can use to test. It does have 32 GB of RAM though. 
I have 2 files that I'm having trouble with. I'll focus on the first here. It looks like parts 1 and 2 finish with no errors or warnings. However, there is no dataset in ms2.out. There are no files in the 'results' folder, not even in the QC folder.
Part 3 shows the error along with warnings. I've attached a copy of the console if that is helpful.

[21072 console.txt](https://github.com/wadpac/GGIR/files/8946572/21072.console.txt)

jmkurka83, 2022-07-11:
Hi Vincent. Following up on this issue. Please let me know how I can help. Thank you!

vincentvanhees, 2022-07-26:
If there is no data in ms2.out then that indicates that GGIR part 2 somehow did not manage to do it's job.

**Inspect RData file from part 1:**

Load the content of the .RData file stored in meta/basic and try to see whether it looks  plausible.

`print(nrow(M$metahort))`

gives the number of epochs extracted from the data. If this is low then there was not enough data in the recording.

`print(summary(M$metahort))`

gives as a summary of each of the columns and helps to check whether value range is plausible.

`(table(M$metalong$nonwearscore) / nrow(M$metalong))  * 100`

gives percentage of nonwearscores: 0 (likely worn), 1, 2, 3 (increasingly likely not worn)

**Inspect file with GENEActiv software directly:**

Check recording length and plausible value range.

**Your GGIR script:**

Another factor could be the way you use GGIR. For example, you use f0 = 70 and f1 = 70. When investigating a problem I normally try to isolate the issue by putting a single file in a data folder and trying to process only that. First with the GGIR default arguments and if that does not reproduce the issue then slowly expanding to represent script you have used.