wadpac/GGIR #565: test_gloadlog fails on OSx

vincentvanhees, 2022-05-24:
As [noted](https://github.com/wadpac/GGIR/pull/561#issuecomment-1129604691) by @tarensanders and now also visible in [CRAN checks for OSx](https://www.r-project.org/nosvn/R.check/r-release-macos-arm64/GGIR-00check.html) [test_gloadlog](https://github.com/wadpac/GGIR/blob/master/tests/testthat/test_gloadlog.R) fails for OSx. This relates to unit test for [g.loadlog.R](https://github.com/wadpac/GGIR/blob/master/R/g.loadlog.R).

It is unclear whether the source of issue is update to R4.2.0 or the code itself. I do not have OSx and somehow I am not able to reproduce the issue in GitHub Actions.

@tarensanders would you be able to test whether the disappears by undoing [this](https://github.com/wadpac/GGIR/commit/128ac3567ac0d77f7b5d990510062063c6c8ec79) commit?

In the meantime I will have a investigate why GitHub Actions does not reproduce the issue.

tarensanders, 2022-05-24:
Hi @vincentvanhees, the tests still fail after reverting that commit.

I ran `git bisect` using f291602fc7bf338ea3e0a26b6c49eaead50889c2 as the last good commit (since those tests passed when I created the PR), and it suggests 891820b6745ce3df2c5dafcb0908441b5a701b6f as the culprit, but I think it's getting confused because there are some tests that fail that are unrelated (like 6b40a6df127bab6ca0dbe868b715fcbf75c47e3c).

I am fairly sure that the issue starts in #544. The commit immediately prior to this (8dcafcca6a89286bf06022f6bf90da2a7cd76d7f) passes, but the merge commit fails. Hope that helps, let me know if you want me to test any changes locally.

Also, I don't think it's related to R version. I am using 4.1.3.

tarensanders, 2022-05-24:
Since that PR includes a change to the way the test file is made, I've attached the version generated by the mac in case you want to compare to the ones on linux/windows. [testsleeplogfile.csv](https://github.com/wadpac/GGIR/files/8762591/testsleeplogfile.csv)

vincentvanhees, 2022-05-24:
thanks

> I am using 4.1.3 on x86_64-apple-darwin17.0 (64-bit).

then it remains unclear why the same tests on the same code pass for OSx on GitHub Actions and not for OSx on your machine or for OSx on CRAN.

tarensanders, 2022-05-24:
I think I know what the issue is.

As backstory, we have a package that was one of ~700 CRAN threatened to archive because one of the dependencies (lubridate) failed a check on OSX (you can see a thread about it [here](https://github.com/tidyverse/lubridate/issues/998)). The underlying problem ended up being a system-specific issue to do with the timezone (see https://github.com/tidyverse/lubridate/issues/991 and the associated fix https://github.com/tidyverse/lubridate/commit/f4a4e426588729c0b781613ddc0fa688aaf29b68).

This makes sense for this problem. Presumably GitHub/CRAN/Me all have different timezones.

As a test, I added `Sys.setenv(TZ = "UTC")` to the top of `test_gloadlog.R` to force the timezone, and all the tests pass. So the issue seems to be how one of the base functions (maybe `as.Date`?) is dealing with timezone differences.

I'm afraid I haven't got a fix, but might be able to work one out from the lubridate changes above.

vincentvanhees, 2022-05-25:
I have run a GitHub workflow ([yml file](https://github.com/wadpac/GGIR/blob/issue565_osx_workflow/.github/workflows/r_basic_check_osx.yml)) that runs the GGIR checks and tests on OSx in either Australia/Sydney or Pacific/Auckland, but no luck yet reproducing the failing tests.

vincentvanhees, 2022-05-25:
> Since that PR includes a change to the way the test file is made, I've attached the version generated by the mac in case you want to compare to the ones on linux/windows. [testsleeplogfile.csv](https://github.com/wadpac/GGIR/files/8762591/testsleeplogfile.csv)

File is identical to the file I generate in Ubuntu, so that does not seem to be the issue.

vincentvanhees, 2022-05-25:
> one of the base functions (maybe as.Date?) is dealing with timezone differences.

I have now expanded every call of as.Date in the g.loadlog function with a specification of tz. I suspect that that was the problem because otherwise it may default to UTC, which would lead to a different date if we would forget to specify tz in one place and not in the other.

https://github.com/wadpac/GGIR/tree/issue565_fix 

Would you mind testing whether this addresses the issue? I will make a PR shortly.

tarensanders, 2022-05-26:
> I ran a GitHub workflow ([yml file](https://github.com/wadpac/GGIR/blob/issue565_osx_workflow/.github/workflows/r_basic_check_osx.yml)) that runs the GGIR checks and tests on OSx in Pacific/Auckland, but no luck yet reproducing the failing tests. Also tried Australia/Sydney but same result.

How odd. Can confirm that setting my TZ env variable to Pacific/Auckland fails. If you still want to pursue this, maybe try printing the timezone as recognised by R as a sanity check that it's actually being changed?

vincentvanhees, 2022-05-26:
> maybe try printing the timezone

Good point, I will incorporate this and will also change the OSx default testing timezone on GitHub Action from Asia/Singapore to Pacific/Auckland. Note that we currently test Windows in US and Linux in Europe. I thought this would be a reasonable compromise relative to testing all OS systematically across all corners of the world.

vincentvanhees, 2022-06-06:
Hi @tarensanders, I see GGIR is still failing on CRAN checks for OSx: https://cran.r-project.org/web/checks/check_results_GGIR.html

It worked fine when you tested it, so it looks like we are still failing to fully address the issue somehow.

tarensanders, 2022-06-07:
Hi @vincentvanhees, how frustrating. Since both the failures are on arm64, I setup a M1 mac instance on [Scaleway](https://www.scaleway.com/en/hello-m1/) but I cannot get those errors to reproduce. `devtools::check()` passes fine, and I can run the tests for gloadlog with no issues. The only difference between the Scaleway instance and CRAN is that Scaleway only offers macOS 12, and [CRAN uses 11](https://cran.r-project.org/web/checks/check_flavors.html#r-release-macos-x86_64).

AWS have a [M1 instance with macOS 11](https://aws.amazon.com/ec2/instance-types/mac/), but it's in preview and I don't have access. I will ask around to see if anyone has an M1 mac with macOS11 to see if I can at least replicate the error. Hard to debug otherwise.

tarensanders, 2022-06-07:
OK, I don't think it has to do with the OS. I can replicate the errors on the M1 instance by setting a timezone in the options (`Sys.setenv(TZ = "Australia/Sydney")`), so it seems like it is still a datetime issue.

I think this is because the function `iso8601chartime2POSIX` uses the system timezone if an empty string is used. If I try and convert the `rec_starttime` using the system timezone, I get the 26th. Is that intended?

Also, as a bit of a sanity check, I added a step in the GitHub Actions to print the timezone (as recognised by R) to the log. As you can see [here](https://github.com/tarensanders/GGIR/runs/6768974587?check_suite_focus=true), both `Sys.timezone()` and `Sys.getenv("TZ")` return `UTC`. So the step in actions to set the timezone doesn't not actually seem to work (or, is ignored by R).

I'm logging for the day, but will have another look at this tomorrow.

tarensanders, 2022-06-09:
Vincent, here is my best guess. 

In the g.loadlog test, `rec_starttime` is set with a timezone (`+0200`). That gets passed in here
https://github.com/wadpac/GGIR/blob/33d6fadebbfbf3b65e4e5da8ac5d9c46118b0a45/R/g.loadlog.R#L14-L16

When the startdates are set a few lines later, it's converted using `iso8601chartime2POSIX`, which uses the *system timezone* if a blank string is passed in (which is the default).
https://github.com/wadpac/GGIR/blob/33d6fadebbfbf3b65e4e5da8ac5d9c46118b0a45/R/g.loadlog.R#L22

The result is that if the timezone is different by ~4 hours, it shifts the date, and you get this mismatch.

Two things aren't clear to me:

1. Why this passes on systems other than macOS. It seems like this should always fail in other timezones.
2. If it's strictly necessary for `rec_starttime` to contain a timezone. It looks like maybe it is from the code in `g.part3`.

So one possible solution would be to replace `rec_starttime` with `format(as.POSIXct("2016-06-25 20:20:20"), "%Y-%m-%dT%H:%M:%S%z")` to coerce it to have the system timezone.

vincentvanhees, 2022-06-09:
Thanks. I just found a way to let the workflow reproduce the errors on Mac / Pacific timezone: See PR #579.

As you will [see](https://github.com/wadpac/GGIR/pull/579/files) the .yml code has become a bit longer because I no longer use szenius/set-timezone@v1.0, and I have moved the timezone specification to right before the R package check.

Windows and Ubuntu do not fail, probably because they are configured to run in Europe and US, which for 8pm is sufficiently far away to cause date confusion.

I will look at your suggestion in relation to test_gloadlog in the upcoming days/weekend, but it sounds like you identified the issue.

vincentvanhees, 2022-06-13:
just to follow up, I have now committed your suggestion to PR #579.

This PR will then address two issues:

1. The default rec_starttime = "2016-06-25T20:20:20+0200" assumed data to be collected in Europe, while the rest of GGIR uses local time (Pacific in this case). This then resulted in conflicting date observations.
2. The GitHub Actions workflow was not sensitive to timezone specification.

vincentvanhees, 2022-06-13:
... the PR still [fails](https://github.com/wadpac/GGIR/runs/6866389995?check_suite_focus=true) for Macos.

I will be travelling this week, hopefully more time to work on it in the upcoming weeks.

tarensanders, 2022-06-14:
Actually it looks like that did work, it went from 11 failing to 6. I should have been more specific, that same line appears twice more:

https://github.com/wadpac/GGIR/blob/9d47fc12930470d81128bf6a8f8d735cd79df935/tests/testthat/test_gloadlog.R#L26

https://github.com/wadpac/GGIR/blob/9d47fc12930470d81128bf6a8f8d735cd79df935/tests/testthat/test_gloadlog.R#L40

vincentvanhees, 2022-06-21:
Thanks, PR #579 has been merged now.

PR #568 (2022-05-26): gloadlog sensitivity to timezone on macOSx
Attempts to fix #565

@tarensanders would you mind testing this?
closed 2022-06-21 by vincentvanhees