wadpac/GGIR #566: object cl unknown when processing file in serial mode

vincentvanhees, 2022-05-25:
We have  `on.exit(parallel::stopCluster(cl))` but after PR #561 was merged it cannot find object cl do.parallel = FALSE. 

In other words: GGIR can only process multiple files in parallel at the moment. Will need to be fixed in upcoming week + with new CRAN release.

vincentvanhees, 2022-05-25:
... so the problem is that the decision whether or not too process in parallel was made too late in the code.
I have now moved code up such that the decision is made before that decision.

tarensanders, 2022-05-26:
I'm running this at the moment, but I don't think I follow. The line `on.exit(parallel::stopCluster(cl))` is wrapped by the if statement https://github.com/wadpac/GGIR/blob/536ffafee772440638fd14aee70b3f493441a65c/R/g.part1.R#L397, so if you specify `do.parallel = FALSE`, it would never run? Do you mean you can't do a single file if `do.parallel = TRUE`?

vincentvanhees, 2022-05-26:
Yes, the problem was that do.parallel was set to FALSE if there is only a single file to process. 
However, that meant that object `cl` was not created because it happend inside the `if (params_general[["do.parallel"]] == TRUE) {` section.

As a result, `on.exit(parallel::stopCluster(cl))` at the bottom of the function was problematic, because this line assumes that `cl` exists.

In the updated version of the code object `cl` is only created AND used when running in parallel.

PR #567 (2022-05-27): Fix to #566

Attempt to fix #566 The problem is that the decision whether or not too process in parallel was made too late in the code. 
I have now moved code up such that the decision is made before that decision.

@tarensanders would you have time to review this before I merge it in the upcoming days? I think it is important that this is fixed soon and submitted to CRAN early next week when we have hopefully also addressed the timezone issue #565 .

Checklist before merging:

- [x] Existing tests still work (check by running the test suite, e.g. from RStudio).
- [x] Added tests (if you added [... cut]