wadpac/GGIR #573: Code redundancy g.imputeTimegaps and read.myacc.csv

vincentvanhees, 2022-05-31:
Both g.imputeTimegaps (primarily used for ActiGraph) and read.myacc.csv (primarily used for ad-hoc csv formats) try to impute time gaps in the raw acceleration data.
 It would be good if we avoided this code redundancy.

jhmigueles, 2022-07-13:
Great! Should we replace the c(0, 0, 1) for the last known value recorded in the previous chunk of data? That way the imputation method is consistent throughout the process.

And another question, in lines 39:43 of the same function, there is a normalization of the imputed data:
```
#  normalise last known value to 1
      i_normalise = which(x$gap != 1)
      if (length(i_normalise) > 0) {
        x[i_normalise, xyzCol] = x[i_normalise, xyzCol] / sqrt(rowSums(x[i_normalise, xyzCol]^2))
      }
```
Should this be done? The last recorded value is already a plausible value. When normalising it, we get many non-meaningful decimals and the dataset changes a bit (e.g., we get different results from processing an actigraph gt3x and an actigraph csv file). Should we avoid this step?

vincentvanhees, 2022-07-13:
> Great! Should we replace the c(0, 0, 1) for the last known value recorded in the previous chunk of data? That way the imputation method is consistent throughout the process.

Which line are you referring to? If it is the first sample in the recording then there is no previous chunk of data. So, make sure this also works when processing the first chunk in the recording.

> Should this be done? The last recorded value is already a plausible value. When normalising it, we get many non-meaningful decimals and the dataset changes a bit

I did this because if there is large time gap then gravity must equal to 1 during that entire period (complete absence of movement). If the last recorded value does not have vector 1 then that is a plausible for imputing a longer time period.

>  (e.g., we get different results from processing an actigraph gt3x and an actigraph csv file).

Where do the differences come from? If it is an inconsistency in our own code then we need make this consistent such that both use the normalised value.

jhmigueles, 2022-07-13:
> Which line are you referring to? If it is the first sample in the recording then there is no previous chunk of data. So, make sure this also works when processing the first chunk in the recording.

Yes, in case it occurs in the first chunk of data, then c(0, 0, 1) is still used.

> Where do the differences come from?

Regarding the normalization, what I've found in my data is that the last epoch is always very close to 1 (e.g., EN=0.999585), which likely relate to calibration error. Actilife exports the csv dat with these gaps already imputed (copying the last recorded value). GGIR does not identify the idle sleep periods in the csv files because they are already imputed. So, if I process the csv file and the gt3x file of the same recording, the csv would not be normalised while the gt3x would be normalised.

vincentvanhees, 2022-07-13:
ok, I see the problem. I think what we do is easier to justify.

I see two solutions:
1. Contact Actigraph and ask them to update their software.
2. Update GGIR as follows: Impute by last value => Check whether vector is less than 0.005 _g_ different from 1 _g_ (5 m_g_)
=> If yes, then normalise => If not, keep as it is. This assures that large errors are addressed.

jhmigueles, 2022-07-13:
Ok, I will contact ActiGraph and implement the point 2 as a temporary solution, does it sound good?

jhmigueles, 2022-07-20:
An update on this: I already solved the code redundancy and everything work fine. I was testing it in different files and found something I would like to discuss. When calibrating the signal, g.imputeTimegaps is used on ActiGraph csv files ro remove the zeros in the file in order not to affect the calibration. Now that GGIR uses the last observation carried forward, g.imputeTimegaps does not find the zeroes and the periods with the idle sleep mode are used in the calibration process. To be consistent with the gt3x files or any file with time gaps in which these periods are not imputed, we could:

1. Find periods with repeated values for more than 1 second in actigrap csv files and remove them to be reimputed in GGIR (g.calibrate would not re-impute them). I have tried this and in my files the calibration process run the "recalibration attempted but possibly not good enough" message (example file: cal error before = 0.02152, after: 0.00375).
2. OR reimpute the timegaps prior calibration with g.imputeTimegaps, but I'm not sure if this approach is reasonable. I have also tried this, and g.calibrate run the "recalibration done, no problems detected" message (example file: cal error before = 0.0171, after: 0.0046)

vincentvanhees, 2022-07-23:
My understanding is that g.imputeTimegaps does not impute the time gaps in g.calibrate.R, because argument impute = FALSE. It should only removes zeros. Therefore, I am not sure the data are imputed before it arrives in the g.calibrate. Within GGIR the time gaps are only imputed in g.getmeta.R, but that is after the calibration procedure. The scenarios where we will encounter imputed raw data in g.calibrate is when the data is exported to csv from Actilife or for some ad-hoc .csv format data.

So, if you want to work on this then I think we can limit our effort to .csv data only. Agree?

A possibly faster solution is to let GGIR just search for all non-movement periods in the data as it is, and then at the end of that process we check whether many of the calibration sphere data points are identical (not similar but exactly the same for all decimal places). If we find many duplicated points then we can remove them. Computationally this may be faster then scanning the raw data for idle.sleep.mode and again for non-movement periods. To avoid risking to bias the data from other sensor brands we could limit this step to .csv data only.

jhmigueles, 2022-07-28:
I have been looking at this today, I applied the two possible solutions, below the results. 

In brief, the current GGIR master github branch produces quite different calibration output if loading a gt3x or an actigraph csv (exported from actilife with the last observation carried forward). If we try to detect when the idle sleep mode has been activated and remove these points from the raw signal, we get very close calibration output to the gt3x file (but it takes longer, +6.5 vs 2.8 minutes). If we remove duplicates from the sphere as you suggested, the calibration output is still close enough to the gt3x file (and the processing time is not that different, 3.8 vs 2.8 min).

Any preferred approach?

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/jaihid/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/jaihid/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#0563C1" vlink="#954F72">

Calibration   strategy | File format | Parts 1:2 time   (min) | cal.error.start | cal.error.end | n.10sec.windows | n.hours.considered | QCmessage
-- | -- | -- | -- | -- | -- | -- | --
Current   approach, github master branch | gt3x | 2.8 | 0.0126 | 0.0024 | 2788 | 84 | recalibration done,   no problems detected
Current   approach, github master branch | AG csv | 3.3 | 0.0087 | 0.0035 | 21385 | 84 | recalibration done,   no problems detected
Try to   identify idle time and remove it before calibration | AG csv | 6.5 | 0.0119 | 0.0023 | 2148 | 74 | recalibration done,   no problems detected
Remove   identical points in the sphere when calibrating | AG csv | 3.8 | 0.0105 | 0.0027 | 1196 | 84 | recalibration done,   no problems detected

</body>

</html>

vincentvanhees, 2022-08-09:
Sorry for slow reply. In that case let's go for option 2:

> If we remove duplicates from the sphere as you suggested, the calibration output is still close enough to the gt3x file (and the processing time is not that different, 3.8 vs 2.8 min).

The differences may then come from the fact that GGIR tries to read 72 hours in total, which are different dependent on whether timegaps are included or excluded.

vincentvanhees, 2022-08-09:
Heads-up: I just merged master branch into branch issue573_imputeTimeGaps.

vincentvanhees, 2022-08-09:
@jhmigueles I am going through your progress, nice work. A few comments:

1. You used `accelerometry::rle2` in g.imputeTimegaps. Would it be possible to replace this by the normal rle() function from R base? In that way we avoid having to add another package dependency.
2. I see you are exploring an alternative imputation approach with zoo::na.locf(x). If it works then good, but please keep an eye on processing speed. The way I implemented it was primarily because I could not find an alternative way to do the computation fast.
3. In files with many and/or large timegaps GGIR will struggle with memory. So, once the work in this branch is merged with master I think the next step could be to modify the code to skip gaps longer than 90 minutes, and only mark them for imputation at the metric (epoch) level. For example, we could do this by passing on the x$gap column to g.getmeta.

jhmigueles, 2022-08-11:
Thanks @vincentvanhees!

1. I tried this with the base rle function but I couldn't do it. Anyway, if we go for option 2 as you suggest we do not need the `accelerometry::rle2`. Sorry I have this option 2 only in a local branch in my computer, I will get it in github today.
2. Great, I will test the processing speed for that chunk of code in master and this branch and will use the fastest.
3. That sounds great, this will be the next step.

PR #612 (2022-08-15): Issue573 impute time gaps

- Fixes #573 
- Additionally, when reading ActiGraph csv files in previous versions, some lines of the csv's were ignored. This was because in `data.table::fread`, the argument header was set to TRUE (default), making that in each iteration, the first row of acceleration values was considered a header (and then skipped in the resulting data frame).
- Also, g.imputeTimegaps is now extended to consider potential time gaps at the begining of the file, time gaps at the end of the file, and potential time gaps that start in one iteration and finish in the next one.

Check [... cut]
closed 2022-08-15 by vincentvanhees