wadpac/GGIR #580: Gt3X time zone enquiries

angerhang, 2022-06-13:
At the moment, GGIR assumes that the start time read from `info.txt` is the true local start time because GT3X doesn't have time zone information. The start time however in `info.txt` defaults to UTC as described here.

https://github.com/actigraph/GT3X-File-Format/blob/main/LogRecords/Parameters.md

So if a data is collected in a non-GTM time zone, what we should really do is to use the `UTC_OFFSET` from the PARAMETERS or `time_zone` from into.txt to correct the timezone. Otherwise the time zone and day light saving information is going to be wrong. There is a related discussion seen in the Google group already:

https://groups.google.com/g/rpackageggir/c/7WopOIbAHRs/m/ya-Tm0mTAgAJ

Would be great to get your views on this @vincentvanhees  :D

angerhang, 2022-06-13:
> > There is a related discussion seen in the Google group already:
> > https://groups.google.com/g/rpackageggir/c/7WopOIbAHRs/m/ya-Tm0mTAgAJ
> 
> Yes, that google group question resulted in a file exchange outside the group. I then used that file to work on issue #551 via PR #557, which as you can see is still open. As you see in the PR discussion the issue was almost fixed until we found some odd discrepancy in behaviour between operating systems. Either way, the idea was to ask the user what timezone they want the data to be interpreted in, because in GGIR we already do that for other data format. However, I agree that it would be nice if this defaults to a plausible value when no desired timezone is not provided.
> 
> > So if a data is collected in a non-GTM time zone, what we should really do is to use the UTC_OFFSET from the PARAMETERS or time_zone from into.txt to correct the timezone. Otherwise the time zone and day light saving information is going to be wrong.
> 
> Yes that makes sense and sounds in line with Matt's recent [suggestion](https://github.com/wadpac/GGIR/issues/551#issuecomment-1142578529). My only doubt has been whether I should make an effort to get this fixed inside the read.gt3x R package such or first to try to address it in GGIR only.
> 
> If we first focus on GGIR then I think best steps are:
> 
> 1. Create issue in read.gt3x repository to inform maintainers that it is an issue and that it is being looked into.
> 2. Follow your suggestion to extract the information from the info.txt.
> 3. Merge this work with PR [Allow for specifying desiredtz and configtz when working with gt3x files #557](https://github.com/wadpac/GGIR/pull/557), because I think it would be nice to combine this develop effort with fixing that issue.
> 4. Test that it works on at least Windows and Linux.
> 
> I am travelling this week, but can spend a bit of time on it later this month/summer. Would you be interested to help develop and/or test this?

Hey Vincent your suggestion looks very sensible to me. That's probably what the GGIR's python sister package should also follow. I would be willing to help. However, my knowledge of R is limited to usage only probably not good enough for development sadly.

I would be happy to contribute to the discussions and testing side of things as I am familiar with the GT3X file format and have developed Python version of the gt3x reader. 

Just give me a shout out whenever you think things are ready for testing! I would then compare the output format from my end accordingly.

angerhang, 2022-06-20:
Actigraph seems to be working  a Python version of GT3X reader now. https://github.com/actigraph/pygt3x It is fairly recent and its features are limited but good to keep an eye on this.

vincentvanhees, 2022-08-10:
Sorry, it took me a while to find time for this. I think I fixed it, see my elaborate response here: https://github.com/wadpac/GGIR/issues/551

I will close this issue now, as well as the old draft PR.

closed 2022-08-10 by vincentvanhees